vedderb/bldc #316: Cannot self-compile firmware

dakhnod, 2021-07-09:
I tried uploading self-compiled firmware to my VESC6-MK5, but the VESC always fails to boot
with a short green flash, a red led flash and a reboot, endlessly looping.

Steps I performed:
1. Cloned original bldc repo
2. installed newest arm-toolchain
3. ran `make`
4. connected VESC tool to VESC and uploaded the custom file `build/BLDC_4_ChibiOS.bin`
I left config_general.h untouched, since it looked pre-configured for HW 60 MK5.

I also tried uploading the .bin via jlink to address 0x0, resulting in the same bootloop.
Uploaded VESC_default aswell, which unbricked my controller.
But uploading the self compiled bin always resulted in the booltoop...

Is there any configuration I am missing?
How should I compile the code to work with my VESC6 MKV?

Peemouse, 2021-07-10:
Hi,
You have to use a soecific version of the toolchain.
Please refer to the Readme of this repo

dakhnod, 2021-07-10:
@Peemouse thanks, your are my saviour

closed 2021-07-10 by dakhnod