vedderb/bldc #319: Use realitive instead of absolute position for pid position control

Teslafly, 2021-07-14:
I see you have somewhat recently reworked/touched pid position control, so now may be a good time to bring this up.
Right now, pid position control normalizes all pid angles to a positive value 0-360 degrees. 

This leads to some issues where the motor tries to turn around completely when reaching the end of the 0-360 range, and also means that "position angle division" setting to make 360 degrees = 2+ turns does not work.

My application is using the vesc like a servo with an attached output gearbox. I need to turn the motor -1800 to +1800 degrees (4 turns) after setting a zero point with an endstop. (call: mcpwm_foc_get_tachometer_value(reset=True)) 

I have made a branch that implements this pid position behavior at https://github.com/Teslafly/bldc/tree/gearbox_servo_app_dev
See my changes to mcpwm_foc.c
I am using the as5047 spi hall encoder to track the motor position which gives good results.

### So basically:
`mc_interface_set_pid_pos()` should not use `utils_norm_angle(&pos);` 
You should be able to send a positive or negative value of any angle, even over 360 degrees, and be able to treat the vesc as a servo going to those values. 

"position angle division" > 1 should act as an angle multiplier, with "position angle division" = 7 making the motor spin 7 degrees for each 1 degree of command. (that, or if the angle is meant to be divided by it, the behavior is just inverted.)
This also removes any need for special behavior based on the value of `p_pid_ang_div`. As it is always tracking the motor in a relative way.

I am wondering what your thoughts are on this behavior change? Is there a reason you made it absolute originally? If we switch to relative behavior, do you want to do it? or I can make a pull request. It may take me a little while to update my method though.

supercrazysam, 2021-07-15:
Absolute positioning is very useful for robotics application. You can use absolute position value to do relative task, but using relative position to do absolute task would be really troublesome. VESC's native support for absolute encoder saved a lot of my projects, honestly I prefer absolute positioning behavior a lot more.

Also if you are looking for "multi-turn absolute" I saw and used a mod somewhere that does absolute positioning multi-turn.
You may want to look into how he changed the encoder.c
https://github.com/raess1/vesc-FW

I followed his mod and successfully implemented multi-turn absolute position control a while back, and used it on many of my projects over the years, you can look at how I modified the code:

https://github.com/supercrazysam/bldc_speed_custom
https://github.com/supercrazysam/bldc_speed_custom/commit/23d7f27bd9b71d604eef9c947361f699beff2060

supercrazysam, 2021-07-15:
Also the author of the multi-turn absolute positioning mod,  raess1, submitted a pull request to add the multi-turn behavior a while back, and unfortunately didnt get through.
https://github.com/vedderb/bldc/pull/65

Teslafly, 2021-07-15:
so "absolute" position control is actually what i'm looking for. It was probably a bad idea to reference "absolute" as the angle being normalised to 0-360 degrees positive only. (see `utils_norm_angle(&pos);`).

#65 seems like what I want. The killer in my application was the one fixed by "2: Removed shortest route rotation -+180 degress". which is caused by `utils_norm_angle(&pos);` and the diff angle util.

If we want to merge #65, it needs to be cleaned up and applied to the dev_fw_5_03 branch. There seems to be a lot of files in #65 that shouldn't be there.

supercrazysam, 2021-07-15:
Yes, glad that the multi-turn approach can solve the single-turn related issue you encountered. 

Also in #65 there are a lot of extra codes not related to multi-turn absolute positioning.  In the past, I went through the code, separated and extracted it to use in my projects. 

I can do it again and submit a purified pull request based on the mod and my tested functioning project code.

But before doing all that, I do want to know the reason behind using "single-turn absolute positioning" as the default option in vesc...

Lets wait for Benjamin's response.

Teslafly, 2021-07-15:
looking at #65, you seem to have done a similar thing that I did in https://github.com/Teslafly/bldc/tree/gearbox_servo_app_dev

when the position is close to the limit and rolls over, increment/decrement the overall angle.

```
// angle now ranges from 0 to 360 degrees
	// update if the motor made a revolution
	if ((angle_now > 300) & (motor_now->m_pid_div_angle_last < 60)) {
		motor_cumulative_revolutions -= 1;
	}
	if ((angle_now < 60) & (motor_now->m_pid_div_angle_last > 300)) {
		motor_cumulative_revolutions += 1;
	}
	motor_now->m_pid_div_angle_last = angle_now; // save previous angle

	motor_angle = (motor_cumulative_revolutions * 360) + (angle_now - pid_angle_zero_offset);
	motor_angle = (motor_angle) / (conf_now->p_pid_ang_div);
```

vs 

```
	// Handle encoder rollover up or down.
	if(last_encoder_counts < (16384 / 4) && pos > (3 * 16384 / 4))
	{
		cumulative_encoder_counts -= 16384;
	} else
	if(last_encoder_counts > (3 * 16384 / 4) && pos < (16384 / 4))
	{
		cumulative_encoder_counts += 16384;
	}

  // Now account for actual reading changes
	cumulative_encoder_counts += (pos - last_encoder_counts);
	last_encoder_counts = pos;
```

Except you did it in the encoder code and I did it in the foc loop.
Is there a reason to do it in the encoder? you would need to implement it per encoder then now that vesc supports multiple encoders.

vedderb, 2021-07-18:
There are many applications where there are no obvious end stops and it makes sense to wrap a speed controller around a position controller. Then the speed controller can just take current position and increment in by some amount, normalize that angle and pass that to the position controller. Crossing 360 degrees becomes seamless then. You can move any distance in any direction in a controlled manner by incrementing the position at the desired rate, wrapping around 0 several times as you do so.

This makes sense when you have e.g. a robot, cable camera or hoist where you want to run at constant speed, and the load varies significantly based on position and direction. Setting speed 0 should stand still without requiring an integrator that drifts forth and back around 0 to stand still.

CNC-applications with e.g. step/dir also work well with the wrap around at 180. Any increment to the current position less than 180 degrees will always turn the motor in the same direction as the sign of that increment with an amount proportional to the increment. It actually solves the problem that I believe you think it causes.

In applications with a defined linear range the wrap around at 180 can easily be solved by doubling the position angle division. Even if it wouldn't wrap around at 180 it would wrap around at 360, so it does not really solve the problem either unless you keep incrementing the angle and remove the assumption of rotation. A position state without bounds sounds impractical to me as the number requires more bytes to be stored and there are many mistakes that easily lead to the motor spinning at full speed in one direction without stopping.

If you have an application with linear motion and end stops I would recommend setting position angle division to a large enough value to not get any wrap around, and use mc_interface_update_pid_pos_offset to set to corresponding offset when you hit the limit switches. mc_interface_update_pid_pos_offset tells the position controller to add an offset to the angle such that the current angle becomes the angle you give it as an argument.

supercrazysam, 2021-07-18:
I see, that makes a lot of sense to me. Especially the "normalize angle" part.

`Setting speed 0 should stand still without requiring an integrator that drifts forth and back around 0 to stand still.`
I also have seen many poor implementations that suffer from this exact thing...

The hint of using `mc_interface_update_pid_pos_offset` for limit switch's "zeroing" purpose is really useful!

Ncerzzk, 2021-10-31:
> If you have an application with linear motion and end stops I would recommend setting position angle division to a large enough value to not get any wrap around, and use [...]

While there is a problem, for example after I finished installing my linear motion machine, and I got the pos_offset which was 350°. It means that the move range is -350° ~ 10°。 

There are 2 ways. One is to reinstall the machine to ensure the p_offset is enough small.  The other is to increase the  angle division to large enough, in this case, the range from -350 - 0  is wasted totally, which is not a smart idea I think.

vedderb, 2021-10-31:
This is precisely what the position offset is used for
[image]
You can, for example, move the linear machine to the beginning, set Pos Now to 0 in the calculator and click Apply. It will then calculate Position PID Offset Angle such that giving the PID controller position 0 will move it to that position. Another way would be to move it to the center and set Pos Now to 180. Note that the Position Division Angle has to be set before.

Internally the controller might wrap around when it runs, but as it is circular you won't notice that.

Ncerzzk, 2021-10-31:
Thanks for detail reply, now I got it.

closed 2022-04-19 by vedderb