vedderb/bldc #324: Anybody looked into playing tunes on VESC?

supercrazysam, 2021-07-20:
As mentioned in the title.

Many commercial BLDC ESCs produce a start-up tune.
Have anyone tried this on VESC? Would like to know more on this.

Sam.

surfdado, 2021-08-04:
VESCs don't have any built-in beeper/buzzer so this is not possible without a hardware change or the addition of external modules

Addy771, 2021-08-04:
@surfdado it's possible without any hardware changes, only software. Modulating the PWM frequency would work, think about the sounds VESC makes during motor detection or the beeps that hobby ESCs make using the motor.

surfdado, 2021-08-04:
The sounds I hear on my drone and other ESC based devices come from the motor? Not the ESC? I had no idea

supercrazysam, 2021-08-05:
> The sounds I hear on my drone and other ESC based devices come from the motor? Not the ESC? I had no idea

I think so.. just not quite exactly sure how they do it, I saw people use stepper motor+driver to produce music tunes, there are some people did that with bldc motors+driver as well

supercrazysam, 2021-08-05:
My guess is controlling the motor to produce vibration frequency in audible range??
Does anyone have a idea how they actually do it?

arpan0193, 2021-08-05:
Maybe add some soundwave signal to the reference signal?

hexakopter, 2021-08-05:
> There was a video I saw a couple years ago where someone had programmed the ESC for their scooter's brushless motor to play music as it powered the scooter along.

I think you are talking about this video: https://www.youtube.com/watch?v=7BQ9lei_24Y 
You can also read a little bit about it on Shanes website: http://scolton.blogspot.com/2010/12/midi-scooter.html

There was also a request on the vesc-project website for a "beep" implementation a few years back. (https://vesc-project.com/node/647) Maybe with the recent HFI implementation some fundamentals are build into the VESC code that can be used to utilize the motor as a speaker more easy.

supercrazysam, 2021-08-05:
Guys I found out how to do it. Just set the switching frequency to (audible frequencies /2).

Currently writing a prototype to test out the beeping function. :D

supercrazysam, 2021-08-06:
Submitted a pull request on this, it works really well. Unfortunately I don't have my phone right now, I will upload the demo video once I return home.

supercrazysam, 2021-08-06:
Here is the video (using a small drone motor):
[image]

supercrazysam, 2021-08-06:
Updated version:
Demo Video (small drone motor):
[image]    

Demo Video (6" inch direct drive scooter motor):
[image]

supercrazysam, 2021-08-08:
Based on the pull request #328 I made earlier, I was able to made a startup tune prototype.
This is the result. 
[image]
(I tested and it worked on both small drone motor and my 6" direct drive scooter hub motor.)

I will wait for Benjamin to approve #328 first, make necessary changes (if any), then submit this startup tune feature as a new pull request.

closed 2021-08-13 by supercrazysam