vedderb/bldc #331: PAS problem: long engine stop after stopping pedaling

livello, 2021-08-13:
@nitrousnrg 
[image]
[image]
[app_motor_config_livello.tar.gz](https://github.com/vedderb/bldc/files/6980339/app_motor_config_livello.tar.gz)

Unreliable motor stop after stop pedaling. Every time when stop pedaling motor stops very slowly. I think this is wrong way.  Motor must stop immediately after stop pedaling.

Moreover, the motor can continue to rotate for a long time after stopping pedaling.

Sometimes the engine spontaneously turns on even when there is no pedaling

_Originally posted by @livello in https://github.com/vedderb/bldc/issues/243#issuecomment-898224016_

nitrousnrg, 2021-08-13:
in my local branch I have a slight improvement of this code. I have a code that keeps driving the motor at minimum milliAmps for 3 seconds to keep the chain under tension but I don't remember if that code is here.

The improvement was to only count the 3rd PAS pulse going in the right direction, otherwise mild mechanical shocks that make the crank arm go back and forth quickly would be counted as fast forward motion. This solved the spontaneous motor drive.

I keep this local branch as tidy as possible so it can be merged, but its a big code contribution with new comm protocols that has some untested or sloppy code mixed in, its a headache to align everything for the pull request but its high on my list.

livello, 2021-08-14:
> big code contribution
I started to study your code. Great style, good variable and function names. In general, everything is clear. And even I figured out the complex formula for calculating the stop delay.
My pass sensor is a factory made by Panasonic 2gang engine.
The PAS-sensor worked perfectly with the controller TSDZ2 and OS firmware. But I've burn several of them and I switched to VESC. 
I hope, it will last for a longer time )

livello, 2021-08-15:
@nitrousnrg can I use your test build on my ESC?