vedderb/bldc #339: Brake current: even 20A brake current still allows my board to roll downhill...

surfdado, 2021-09-08:
Am I misunderstanding how brake current is supposed to work? I assumed that even a 10A brake current would be sufficient to keep my board from rolling down a 10-15 percent grade. In my experience increasing the brake current above 5A has no added benefit (using a Phub188 style hub motor on a Onewheel).

The motor should be powerful enough to prevent the board from rolling down a 45 degree incline... how do I achieve that?

supercrazysam, 2021-09-10:
Braking depends on two factors in terms of vesc settings, "Motor Current Max Brake" and "Battery Current Max Regen".
May I ask what is your "Battery Current Max Regen" setting value?

surfdado, 2021-09-10:
You're asking about the limits in the motor configuration? They're -100A for motor and -40A for battery, but those are just the limits. Like I said, my motor/setup can easily support my weight on any of those hills. But as soon as I step off and the so called "brake current" takes over the board starts rolling downhill. Slower than without brake current, but still - it rolls.

10amps of motor current should be more than enough to let the 25pound board climb uphill without a rider so I don't understand why 10A brake current can't keep the board locked in place

surfdado, 2021-09-10:
And just to be clear, by "20A brake current" I mean the setting in the balance app, which results in a call to "set_brake_current(..)" in mc_interface as soon as I step off my board:

mc_interface_set_brake_current(balance_conf.brake_current);

vedderb, 2021-09-15:
The brake command only does regenerative braking and not active braking. That means as the speed approaches 0 the braking current also approaches 0. Doing active braking around 0 is more complicated than expected - just applying current in the other direction will make the motor change direction when it stops (and assumes you can track the motor well through 0). To prevent that you have to monitor the speed precisely around 0 speed while the motor is driven, which is close to impossible with anything other than a high resolution encoder. Open loop braking is also possible, but wastes power is tricky to do smoothly. The smart reverse is an attempt at dealing with this problem without any encoder.

closed 2021-09-15 by vedderb