vedderb/bldc #343: Feature Request: Gain scheduling of current, velocity and position gains

Aquaharmonics, 2021-09-28:
Something that would be a great feature would be the ability to implement gain scheduling for the current, velocity, and position gains. If there were a user defined table of gains that could be changed on the fly with an input to say, ADC3, for a user specified range of voltage input the drive would update the gains on the fly. 

This would be useful, for example, motors using a sequential gearbox with an indexed switch that output a different resistance/voltage to ADC3 for every gearshift, that then updated the gains based on the specific gear that has been selected. Additionally, being able to have different ramp rates and current limits for say, acceleration and deceleration within the table for each gear would be useful.

Some PAS (I am thinking specifically of the Bafang BBS/BBHD series) can set torque and speed limits with different PAS levels, and in a similar fashion this could be implemented.

Aquaharmonics, 2021-09-30:
Additionally, this same kind of feature could be used as user input "modes" for say, different power levels/limitations, from say, a handlebar mounted switch.

livello, 2022-04-17:
@Aquaharmonics very nice idea. But i do not understand relation between selected gear and motor settings. There is no such things in Bosh, Shimano, Brose central mtb ebike motors. 
At now we do not have nice working PAS code in master branch. I've corrected this more than 6 months ago and no one did not faced with such problem. 
I think vesc is not common used for PAS on becycles...

Aquaharmonics, 2022-04-21:
@livello, I will try and describe the use case a bit further here. I think this link can help describe the specific problem trying to be addressed so is a good reference for further discussion: [https://www.motioncontroltips.com/when-to-use-gears-to-change-the-inertia-ratio-of-a-motor-driven-system/](url) 

From the link, the main talking points are:

_"Adding a gear set or a gearbox to a motor-driven system reduces the load inertia by the inverse square of the gear ratio, meaning that even a relatively low gear ratio can have a significant effect on the inertia ratio."_

[image]

and

[image]

First I will say that this becomes a very application specific feature, and direct drive systems with a fixed gear ratio wouldn't necessarily needs this feature entirely, but part of the feature implementation would be like the user "profiles" in the VESC mobile app that currently exist, in that different settings could be implemented for the controller and instead of using the app/touch screen interface to go between different profiles, an external switch could be used(handlebar mounted, maybe like the Bafang BBSxx series PAS level switch). **Something I am not entirely sure of is if things like the current gains or ramp rates can be adjusted real time, or if these parameters would have to involve a disabling of the drive, change in parameters, and then a re-enabling of the drive.**

In general, any change to the gear ratio of the system acts like a gain itself, so that the inertia, damping & spring of the system as seen at the motor is effectively changed as the inverse square of the gear ratio. For bicycles, these gear ratios ranges could be something like 300%, more or less depending on the setup. For a sequential gearbox derived from say, shifter karts or motorcycles, the gear ratio ranges could be similar, say 200%-300% depending on donor gear box used. 

If the setup is using current control for instance, say, on a shifter kart with 300% overall range and six gears, if the system was tuned initially in third gear (say mid-range), the same current controller gains used in first gear would be effectively increased by a factor of 2.25 , and for 6th gear would be reduced by a factor of 0.44. (this is a simplification) . 

For a particular mechanical system, this may be fine with a single set of tuning parameters, depending on the inertia ratio, compliance in the system (from gearbox stiffness, belt stiffness, chain stiffness, system damping etc., associated resonances.) For other systems, it could cause the system to lack performance on the one hand, or become unstable on the other hand.

The concept then is to allow the user to define different control loop gains, current limits, velocity limits, ramp rates, for different selected gears based on an ADC input. In this example with the shifter kart, a switch mounted to the shift drum gives feedback of the different gears selected.

[image]

[image]

It is entirely possible that just having two discrete setpoints would be fine, say the same gains for 1st and 2nd gear, and another set of gains for 3rd through 6th gear. 

In addition to the control loop gains, the ability to change ramp rates (typically found in the app settings) would also be beneficial. 

For a bicycle PAS feature, the Bafang BBS series for example does have gear shift sensors, to allow making a shift with the throttle applied, momentarily reducing power, then re-applying after the shift is made. This is only an incremental type feedback and not absolute, so not a perfect solution. the PAS switch settings however could be linked in a table to different controller gains, current limits, velocity limits, power limits etc. 

This is just a start to the description of the feature, and I will come back with a table of a parameters to change with different feedback from say an ADC input corresponding to different selected gears. 

Another way to get an idea of what this feature will do is by simply taking the current controller gains that the VESC tool determines in tuning and multiplying them by the above factors and seeing how the system performs.