vedderb/bldc #362: Consider keeping `master` branch as the bleeding edge

kubark42, 2021-10-14:
For better or for worse, repo branches other than the default one-- `master` in this case-- carry a de facto stigma of not being active, or not being the focus point for development. As a result, it seems that many people are checking out `master` and doing experiments there, not knowing that it is not the active branch. This confusion leads to additional burden on project maintainers and new contributors.

Once v5.03 goes live, it would be worth considering a change in the way the repo is handled. 

#### Suggestion

1. Tag release versions 
3. Use `master` as the bleeding edge, targeting it with PRs
2. Use branches for isolated public feature development

#### Alternative suggestion
2. Create branch `next`, which is the bleeding edge code
3. Set default branch upon clone to `next`
4. `master` keeps a reputation as the stable, tested branch

I think the first suggestion is the better one, as tagged releases are convenient places to backport bugfixes.

ddosoff, 2021-11-19:
Tag release is worse idea than branch release.

In case of tag release you can not easyly add critical fixed to older releases. In case of branches it's easy.

kubark42, 2022-01-18:
> For now I'm keeping a branch named release_x_xx for each release and do the development in master. Should be easy to find old releases then and backport things if needed.

https://discord.com/channels/904830990319485030/910181652192448532/932314885247893525

closed 2022-01-18 by kubark42

Teslafly, 2022-01-18:
> > For now I'm keeping a branch named release_x_xx for each release and do the development in master. Should be easy to find old releases then and backport things if needed.
> 
> https://discord.com/channels/904830990319485030/910181652192448532/932314885247893525

Is there a vesc development discord server or something?

kubark42, 2022-01-18:
> Is there a vesc development discord server or something?

Unofficial, but @vedderb is known to drop in as his time permits: https://vesc-project.com/node/3284