vedderb/bldc #371: [Suggestion] Create tags for sorting issues/PRs

kubark42, 2021-10-31:
It would be very helpful to have a set of tags which can be used to help in issue and PR triage.

https://github.com/TauLabs/TauLabs/labels are the ones we created. They helped us do all kinds of nifty things, such as 
 - identify tasks which were appropriate for new developers
 - show PR's which were intended for comment and extended discussion, but were not yet ready for merge
 - highlight bugs
 - etc...

Could something like that be done here?

kalvdans, 2021-11-02:
> show PR's which were intended for comment and extended discussion, but were not yet ready for merge

github has a function to mark a PR as draft, we could make use of that.

kubark42, 2021-11-03:
> github has a function to mark a PR as draft, we could make use of that.

I gave that a shot on https://github.com/vedderb/bldc/pull/357. Tell me what you think.

I find that the visual clues that it's a draft are somewhat subtle. A slight difference in the icon's shape and color. Dunno why they didn't go for something more obvious. 

The nice thing is that it keeps the PR from being merged until it is out of draft mode. I think my preferred workflow for that would be to start (almost) all PRs as drafts and then to take them out of draft state once the review is done. This way the commit history can be rewritten, squishing and fixing-up all the review changes.