vedderb/bldc #380: FOC wizard shows false information when run for the first time

MayorBug, 2021-11-16:
Environment:
VESC Tool Windows or Android
Tested FW verison b5.03 build 59 and 51 (and many before)
HW: Stormcore 60D+ and 60D

Steps to reproduce:
FOC Wizard (load defaults: YES)  - confirmation screen shows wildly different and false results for one motor 
Actually correct parameters are applied after the detection and both motors will work
[image]

When doing motor detection again without restoring defaults the confirmation screen shows correct information
FOC Wizard (load defaults: NO)
[image]

Jfriesen222, 2021-11-28:
I have noticed also sometimes it seems to display the old config rather then the updated. In this case it is the deefault values.

vedderb, 2021-11-29:
Just gave it a try on desktop and mobile using USB and BLE, and this does not happen for me. I'm using a dual VESC HD where both units are connected over CAN. Maybe this is a problem with the dual motor implementation on the Stormcore, as the HD has separate MCUs.

Does it happen every time for you or only sometimes?

MayorBug, 2021-11-29:
When doing a clean detection on Stormcore it will happen every time

vedderb, 2021-11-29:
I gave it a try on the Stormcore 60D now with both usb and ble, and this does not happen to me. What I did notice though is that the offsets of motor 1 are applied to motor 2, which makes motor 2 run worse than motor 1. I will look into this more later this week.

vedderb, 2021-12-01:
The offset problem is now fixed, it runs much better for me now. I still can't reproduce the detection information bug though.

MayorBug, 2021-12-07:
Thx for fixing the offset problem, that may have been another issue why one of my motors always consumed more energy and was few deg hotter. 

Anyway that wasn't the main issue here for me. I screen recorded a following clip right after I flashed B75. 

https://mega.nz/file/Qo4BnIpB#NsOXSpIJPC27VJYb_SatOtUDkktXggclx-3CMwjyL0M

The first run is clean from the start and you will see that the detection results will show default values (as Jeffrey said) for the first motor. 
On the second run I don't load default values and both motor detection results seem to be correct. 
On the third run I load default values and the bug shows itself again.

MayorBug, 2022-03-25:
I am still facing this issue, is there any logs I could provide to diagnose this?

[image]

vedderb, 2022-12-02:
I think this is fixed in the latest beta. One thing that could cause this is if something is connected over a different port at the same time polling data. In that case it would send the config back to that port. Now it remembers the port that started detection and sends the config back there even if something else communicates in-between.

closed 2022-12-03 by MayorBug