vedderb/bldc #389: ST-LINK 2.1 and OpenOCD

st0rfrasarn, 2021-12-27:
Hi,

For anyone trying to flash via OpenOCD using ST-LINK 2.1 i had to change the following in the Makefile to make it work:

From: 
openocd -f board/stm32f4discovery.cfg -c "reset_config trst_only combined" -c "program build/$(PROJECT).elf verify reset exit"

To:
openocd -f board/stm32f429disc1.cfg -c "reset_config trst_only combined" -c "program build/$(PROJECT).elf verify reset exit"

Teslafly, 2022-01-17:
I remember having to update the openocd board at some point too. but don't anymore.  I believe the fix was to upgrade the stlink firmware or upgrade openocd.

Are you using the stm32f4 discovery board?

is there any reason to keep this issue open?

st0rfrasarn, 2022-01-30:
Sorry did not get any notification about the reply. 

No, problem went away when i flashed. You can close this 😃

closed 2022-01-30 by vedderb