vedderb/bldc #391: [Question] PWM switching frequency is HALF of what is set

ironman223, 2021-12-28:
I was testing my motor on my hardware. while watching PWM pulses on my oscilloscope I found that the frequency of the 
pulses are exactly half of the value that was set.

I tried to adjust it by changing the timer ARR in the foc initialization and measurement of resistance, inductance and where new values are fed into ARR of TIM1. also changed the calculation of "dt".

Now the motor doesn't run, the measured resistance and inductance value are also incorrect.

What I have to test or change to workout this correctly?

ElwinBoots, 2021-12-29:
Your observation is correct, you get half the frequency you set. I am trying to get this fixed as incorrect things do a lot of harm to the confidence in a product.

Please see:
https://vesc-project.com/node/3278

Can you also comment there?

kubark42, 2022-03-10:
@ironman223 this has now been resolved, see https://github.com/vedderb/bldc/pull/397#issuecomment-1008321268.