vedderb/bldc #403: Release tagging and new dev workflow

Teslafly, 2022-01-17:
I see that you  recently released a new version of vesc-tool and vesc-firmware. 

I would not have seen it if i had not been doing some work on it today and saw that you changed up your branches.

What is your recommended workflow with the new branch structure? 
It looks like development is happening in the master branch now? with release branches?
https://github.com/vedderb/bldc/blob/master/CONTRIBUTING or the project readme 
may be a reasonable place to update dev workflow guidelines.

Then with releases, I just tried it with my fork and it is really easy to make a release. Took me 1 minute to click the "draft a new release" button in the releases page and tag the release branch/commit.
https://github.com/Teslafly/Vesc_Firmware/releases/tag/v5.03

There are several existing issues to do releases already. With relatively high "thumbs up" values for this project.

https://github.com/vedderb/bldc/issues/208
https://github.com/vedderb/bldc/issues/31 (even includes commands for historical releases!)

and an issue for vesc tool too.
https://github.com/vedderb/vesc_tool/issues/90

The release branch is a good start, but tags and actual github releases have enhanced integration with github. That integration allows users to find them easily and also generates notifications to project watchers (179 of us now!) when a new release happens.

kubark42, 2022-01-18:
https://github.com/vedderb/bldc/issues/362#issuecomment-1015557460

I agree that tags would be my preference, but it's a style question so I'm fine however. The important change is that the default branch, `master` in this case, will now be where development happens.

Many thumbs up from me for GItHub releases, but that might be a philosophical issue since that release data is owned by GitHub, instead of being stored in the git database.

Teslafly, 2022-03-01:
Discussion is happening. But this ticket is not relevant to it anymore.

closed 2022-03-01 by Teslafly