vedderb/bldc #422: problem with COMM_FORWARD_CAN

Koxx3, 2022-02-28:
Hello,
I am trying to make a gateway for BLE<>CAN and Wifi<>CAN.
It was working perfectly with 5.2 firmware, but not, something changed in 5.3 and the CAN bus it going mad repeating messages in loop between VESCs.
I can't find what's wring.

Here is 5.2 traces during CAN bus scan :
```
17:22:15.591 > VescCommon::receiveBleDataAndTransmitToVesc : send to VESC CAN
17:22:15.593 >  [NOTICE] : VescCan vesc_can_from_stream  : BLE/WIFI => CAN, command 34d <> 22h (COMM_FORWARD_CAN), length 3, longPacket 0 ==> to VESC 95
17:22:15.594 > [NOTICE] : VescCan vesc_can_from_stream : BLE/WIFI => CAN, Buffer extra short processing
17:22:15.595 > [NOTICE] : VescCan vesc_can_from_stream : extrashort / 22 71 00 (3)
17:22:15.596 > VescCan : Sending 0x0000085F / cntrl = 95d / 5fh / cmd = 08h / CAN_PACKET_PROCESS_SHORT_BUFFER, DLC 5, Data 5f 00 22 71 00
17:22:15.597 > [NOTICE] : VescCan vesc_can_from_stream : exit and clean
17:22:15.598 > CAN : Received EXT  from 0x0000118E / cntrl = 142d / 8eh / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.600 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.601 > CAN : Received EXT  from 0x00000871 / cntrl = 113d / 71h / cmd = 08h / CAN_PACKET_PROCESS_SHORT_BUFFER, DLC 3, Data 0x5F 0x00 0x00
17:22:15.603 > VescCan : cmd = 8 / CAN_PACKET_PROCESS_SHORT_BUFFER
17:22:15.605 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x00 0x00 0x05 0x02 0x55 0x42 0x4F 0x58 
17:22:15.607 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.608 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 7
17:22:15.610 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x07 0x5F 0x37 0x35 0x5F 0x33 0x30 0x30
17:22:15.611 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.612 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 14
17:22:15.614 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x0E 0x5F 0x54 0x59 0x50 0x45 0x43 0x00
17:22:15.615 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.616 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 21
17:22:15.618 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x15 0x4E 0x00 0x37 0x00 0x0E 0x50 0x50
17:22:15.619 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.620 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 28
17:22:15.622 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x1C 0x59 0x58 0x38 0x31 0x20 0x00 0x00
17:22:15.623 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.624 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 35
17:22:15.626 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 3, Data 0x23 0x00 0x00
17:22:15.627 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.628 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 37
17:22:15.631 > CAN : Received EXT  from 0x0000075F / cntrl = 95d / 5fh / cmd = 07h / CAN_PACKET_PROCESS_RX_BUFFER, DLC 6, Data 0x71 0x01 0x00 0x25 0xE8 0x9E
17:22:15.634 > VescCan : cmd = 7 / CAN_PACKET_PROCESS_RX_BUFFER
17:22:15.635 > [NOTICE] : VescCan vesc_can_to_stream : CAN => BLE/WIFI, sending 37 bytes, command 0d <> 00h (COMM_FW_VERSION)
17:22:15.636 > [NOTICE] : VescCan vesc_can_read processed frame #1, type COMM_FW_VERSION
17:22:15.657 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x00 0x00 0x05 0x02 0x55 0x42 0x4F 0x58 
17:22:15.659 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.660 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x07 0x5F 0x37 0x35 0x5F 0x33 0x30 0x30
17:22:15.663 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.664 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x0E 0x5F 0x54 0x59 0x50 0x45 0x43 0x00
17:22:15.666 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.669 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x15 0x4E 0x00 0x37 0x00 0x0E 0x50 0x50 
17:22:15.671 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.673 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x1C 0x59 0x58 0x38 0x31 0x20 0x00 0x00
17:22:15.675 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.677 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 3, Data 0x23 0x00 0x00
17:22:15.679 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:22:15.680 > CAN : Received EXT  from 0x00000771 / cntrl = 113d / 71h / cmd = 07h / CAN_PACKET_PROCESS_RX_BUFFER, DLC 6, Data 0x5F 0x01 0x00 0x25 0xE8 0x9E
17:22:15.683 > VescCan : cmd = 7 / CAN_PACKET_PROCESS_RX_BUFFER
17:22:15.684 > CAN : Received EXT  from 0x0000118F / cntrl = 143d / 8fh / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F 
17:22:15.685 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.687 > CAN : Received EXT  from 0x00001190 / cntrl = 144d / 90h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.688 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.690 > CAN : Received EXT  from 0x00001191 / cntrl = 145d / 91h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.692 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.693 > CAN : Received EXT  from 0x00001192 / cntrl = 146d / 92h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.694 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.695 > CAN : Received EXT  from 0x00001193 / cntrl = 147d / 93h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.697 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.698 > CAN : Received EXT  from 0x00001194 / cntrl = 148d / 94h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.700 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.701 > CAN : Received EXT  from 0x00001195 / cntrl = 149d / 95h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F 
17:22:15.702 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.703 > CAN : Received EXT  from 0x00001196 / cntrl = 150d / 96h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
17:22:15.704 > VescCan : cmd = 17 / CAN_PACKET_PING
17:22:15.705 > CAN : Received EXT  from 0x00001197 / cntrl = 151d / 97h / cmd = 11h / CAN_PACKET_PING, DLC 1, Data 0x5F
[... 469 lines cut ...]
17:28:53.582 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.585 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x0E 0x33 0x30 0x30 0x00 0x4E 0x00 0x37 
17:28:53.587 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.589 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x15 0x00 0x0E 0x50 0x50 0x59 0x58 0x38
17:28:53.590 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.592 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x1C 0x31 0x20 0x00 0x00 0x00 0x00 0x01
17:28:53.594 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.597 > CAN : Received EXT  from 0x00000571 / cntrl = 113d / 71h / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 3, Data 0x23 0x00 0x00 
17:28:53.599 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.602 > CAN : Received EXT  from 0x00000771 / cntrl = 113d / 71h / cmd = 07h / CAN_PACKET_PROCESS_RX_BUFFER, DLC 6, Data 0x5F 0x01 0x00 0x25 0xFE 0x27
17:28:53.604 > VescCan : cmd = 7 / CAN_PACKET_PROCESS_RX_BUFFER
17:28:53.606 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x00 0x00 0x05 0x03 0x55 0x42 0x4F 0x58
17:28:53.608 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.608 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 7
17:28:53.610 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x07 0x5F 0x56 0x32 0x5F 0x37 0x35 0x5F
[... cut ...]
17:28:53.614 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x0E 0x33 0x30 0x30 0x00 0x4E 0x00 0x37 
17:28:53.616 > VescCan : cmd = 5 / CAN_PACKET_FILL_RX_BUFFER
17:28:53.616 > [NOTICE] : VescCan vesc_can_read processed frame #1, type << local >> fill rx buffer, size 21
17:28:53.618 > CAN : Received EXT  from 0x0000055F / cntrl = 95d / 5fh / cmd = 05h / CAN_PACKET_FILL_RX_BUFFER, DLC 8, Data 0x15 0
```

each VESC is responding to the other with the firmware version.
What's going on ?

Koxx3, 2022-03-01:
it seems the master VESC is interpreting the slave VESC reply as an order and not something to forward to the requester.

livello, 2022-04-17:
@Koxx3 can you give me advice how to implement this feature?
https://github.com/vedderb/bldc/issues/473