vedderb/bldc #440: Not possible to connect AS5047 to hardware SPI anymore?

floba, 2022-03-09:
`#define AS5047_USE_HW_SPI_PINS		1` in file conf_general.h seems to have no effect anymore. I found this tutorial https://hackaday.io/project/72497-valify-v2-robot-lawnmower/log/140310-2018-04-06 explaining how to use it. The firmware by https://github.com/tlalexander/bldc is "too old" for the vesc_tool. 

Is there a reason for this feature that it was removed? Will it go back?

I fear I have to scroll back the commits to somehow get it to run. I need the absolute value from it via SPI. I have boards with HW 4.12 and 6.x. Ideally, no changes to hardware should be necessary (e.g. unsoldering filters).

Thanks for any response!

Teslafly, 2022-03-10:
For commutation, the software spi is good enough. There is no hardware acceleration currently anyways to the cpu time is taken when talking to the encoder anyways.

The encoder code just got refactored around a month ago now. 

6.x hardware already has no filters and the encoder should work plugged into the hall port.
for the 4.12 hw, If you want to avoid removing the 3 hall filters, you can redefine the hall pins to be the hw spi pins in hwconf and use them as sw pins. you may need the undefine other uses of those pins in other places though.

But eventually more functionality for encoder communication is planned.

floba, 2022-03-11:
Thanks a lot! I got the it run (i.e. read out the encoder via hardware SPI with `#define AS5047_USE_HW_SPI_PINS 1`) by going back to the last commit of Dec 2021 of both bldc and vesc_tool.