vedderb/bldc #451: Compiling under Ubuntu results in 512kb file? unusable

lj502, 2022-03-17:
When I try to compile using the instruction provided here I get a bin file that is 512kb and can't figure this out, any suggestions for me.

John

vedderb, 2022-03-17:
The master branch has a different linker script that will produce 512kb files now to fit more code. You need the latest beta vesc tool and the bootloader with compression support to upload them using vesc tool.

lj502, 2022-03-17:
Thank you that was super fast response, I'm running 5.03 now but need to use the pwm input as the reverse button but my app mode isn't included in that, I can change the code but how do I compile correctly for 5.03
Thank you 
John

Teslafly, 2022-03-31:
you have to check out the release_5_03 branch to get the older version of the code.
https://github.com/vedderb/bldc/tree/release_5_03

If this solves your issue make sure to close this ticket.

closed 2022-04-30 by lj502