vedderb/bldc #463: ADC app does not fail safe when input out of bounds

Teslafly, 2022-03-31:
In the vesc discord, it was noted that pulldown resistors on the adc input do not fail safe in the "current reverse center" mode.
When the throttle is disconnected and the adc falls to 0v, the vesc goes full reverse.

Since hall throttles generally have a non zero minimum voltage ~0.5v if we fail into the `mc_interface_set_brake_current(timeout_get_brake_current())` mode when the input is outside of config.voltage_start / config.voltage_end plus some tolerance, we can fail safe in modes where 0v would otherwise be a throttle input.

I propose 3 new adc app settings:

- bool: voltage_limits_enable (Enable ADC1 failsafe limits)
    - default: true
    - enables limits to disable drive/enable brake.
    - optional. setting stert/end limits to 0v and 3.3v should disable this limit without the bool.
- float: voltage_start_limit (ADC1 min voltage limit)
    - defaults to voltage_start * 0.9
- float: voltage_end_limit (ADC1 max voltage limit)
    - defaults to voltage_end * 1.1

Teslafly, 2022-03-31:
voltage_start_limit (ADC1 min voltage limit)  probably needs to be bounded to a minimum of 0v.

vedderb, 2022-04-01:
Here is an attempt at fixing it: https://github.com/vedderb/bldc/commit/3a530ac3f7b43c06acac807248ae05fef8c1ae97
It is in the latest beta build now too. Can you give me a try and let me know if it works?

Teslafly, 2022-04-05:
As per the discord server, this has been tested successfully. Closing.

closed 2022-04-05 by Teslafly