vedderb/bldc #473: Need advice for new feature: control psu flatpack2 over CAN bus with vesc

livello, 2022-04-17:
There is an electric bike with a 750 Wh battery. I use flatpack 2 48V for charging. I charge on the way several times every 40-60 kilometers. 35A is too much charging current for everyday use.
[image]

It is required to control voltage and current, send commands via the CAN bus. 
It seems to me that the easiest way would be to use the engine power limit value as the charger power target.

There are a lot of such power supplies at a flea market at a price of 50 dollars.

The control protocol for flatpack is known. Here are a few projects that I would like to draw on.
**Tell me how best to implement such an improvement so that it can be included in the upstream.**

Teslafly, 2022-04-17:
Something that this brings up, in seperation from the specific protocol used. Is support for chargers in general. Now that the bms is integrated, having a charger page with charging current and max voltage would probably be reasonable. You could use profiles as well to quickly select what type of charging you want to do. (fast to 80%. Slow to 100%, etc)

livello, 2022-04-27:
@Teslafly thank you! 
I did it in haste. The next step I plan to add is the charge control application and register the calback in the comm_can.c

https://user-images.githubusercontent.com/15374538/165636411-6e7b3e47-a9cd-43a7-812b-f316f111e7be.mp4

https://user-images.githubusercontent.com/15374538/165636433-a31408fe-5a37-47e2-a20a-3f6124c87c0c.mp4

https://user-images.githubusercontent.com/15374538/165636472-84a9ae6f-7cf4-432e-b7a6-07b6087feb74.mp4

https://github.com/livello/bldc/tree/dev_fw_5_03

vedderb, 2022-05-13:
This sounds like something that can be done with lispbm and/or with QML scripting.

Very interesting charger! I just ordered one.