vedderb/bldc #478: App config clears after a reboot or power cycle.

Mitchlol, 2022-04-21:
It looks to be caused by this commit.

https://github.com/vedderb/bldc/commit/349e133b511256c1c6e555b6d198c5de91f15357

Yes, my changes 😭, I figured I'd fess up and put this info here, as it might take me a while to figure out how exactly I broke it.

Last time I broke it, Ben found the issue and fixed it, but iirc it had to do with a bug happening as part of setting/starting/stopping the app config which happens before the write to disk, and prevented it from writing. Seems likely it is something similar this time.

vedderb, 2022-04-21:
Are you sure? Does not seem to happen to me after a quick test. Can you try to run clean before rebuilding?

Mitchlol, 2022-04-21:
A clean fixed it thanks & sorry! 🙌

Not sure how I managed AB test every commit for the last week, but a clean fixed it 🤷

closed 2022-04-21 by Mitchlol