vedderb/bldc #492: Brake current applied on small forward inputs

Thunkar, 2022-05-09:
# Setup

Maker-X DV6
FW 5.03
Fresh flash, correct detection parameters using the wizard
PPM app control, TC enabled
Flipsky 6374 190kv sensored motors
Current control mode
12S LiPo

# Steps to reproduce
When applying small inputs (just above deadband values, in my case 11-12% for a deadband of 10%), the VESC applies a small (-3A) braking current even though the remote is commanding a positive value. Above 13% the current abrutly changes direction to positive and the motor starts spining correctly. This generates a jerky low speed behavior.

This behavior seems unnafected by changing the deadband, control schema (smart reverse also exhibits the problem), redoing detection, measuring offsets ar changing the sensorless preset to <5.02. 

I can also reproduce the issue when attempting to control the motor via keyboard using very low current limits (3A). Even though I'm commanding the motor to go forwards, it will randomly brake and not start spinning.

[image]

# Intended behavior

Current should ramp up smoothly with the PPM input and always be positive in the positive direction (approx 1500ms to 2000ms) and vice-versa.

Thunkar, 2022-05-12:
To add more information, I managed to reproduce the issue on FW 5.2 and 6.0.

The only way I can make it less noticeable under load is to measure offsets, THEN perform detection. Playing with observer gain or other settings caused no effect, but according to my findings it seems to be a problem with either FOC detection or driving this particular motors.

When the issue is exhibited the motors make some weird noises at low duty cycles, like slight cogging. Once they get to ahigher duty cycles, they sound smooth. Furthermore, current consumption on low duty cycles is unusually high (when unloaded) and drops dramatically again at higher duty cycles.

Thunkar, 2022-05-13:
A little bit more experimenting has revealed that lowering the 'Sensorless ERPM' value from the default reduces the time the motor sounds funny during acceleration. I then went on and deactivated the hall sensors altogether and went for HFI, which seems to solve the issue and work a lot smoother after resolving ambiguity.  

Does this hall table look fine? I thought so:

[image]

I'm pretty sure now that the problem is during sensored operation, since after the transition all is fine. Could it be related to https://github.com/vedderb/bldc/issues/139? I find it so weird since the motors are brand new and nothing special.

trisuliswanto, 2022-12-19:
I also have the same problem with the ADC app,