vedderb/bldc #496: Your file directory is too messy

zouzhe1, 2022-05-14:
A project should have a clear file structure. You all put a pile of source files under the root directory, which is very unreadable

Teslafly, 2022-05-14:
While this issue could probably have been put more eloquently, I do agree.
It would be nice if the amount of base files fit on one page of a file tree sidebar.

I think you could get there by placing all root c files in a src folder. Maybe moving motor controller files into their own folder as well (mcpwm, foc_math, mc_interface, virtual motor)
Doing that would also provide a place to seperate out hardware abstraction in the future.

Teslafly, 2022-05-24:
This has been fixed somewhat. Several functional blocks have been moved into their own folders and there are now many less files to wade through in the main directory.

Without a larger restructuring, this is about as good as it will get and this issue should probably be closed.

closed 2022-05-24 by zouzhe1