vedderb/bldc #506: Feature Request: ADC1 mapped for analog breaking and for analog throttle

1ofko6, 2022-07-09:
I would like to add something like a throttle gestures. I explain:
When I twice or [three times (for safety..)] (in short time) press throttle from zero to max, throttle input will act as a brake controller, than I can lineary brake (and regenerate to baterry) when I ride from a hill.

Than, the same gesture to put the input back, in to throttle controller mode.

I was thinking I could program this in scripting, but im not sure if I can control these things from script.. 
In case I could.. withou documentation Im not able to program it...

Thank you for response

geofrancis, 2022-07-09:
that sounds incredibly dangerous, why not just use the first 25% of travel as the brake and the next 75% as throttle?

1ofko6, 2022-07-09:
I dont see any danger there.
I want to use full range of the throttle controler for braking...

Than after braking from hill the brake will have no thet dangerous effect so i will just double click and change it to the throttle mode and then I can go on another hill.

I wanna change the braking current by that lever while riding down the hill.

It oviously should be as a "check box" in the VESC tool.. or etc....

1ofko6, 2022-07-09:
Of course... because many controllers have just one spring and no mechanical center point...
I just go from 0% to 100%, no other option....

Teslafly, 2022-07-09:
Would a button work for changing modes? Beacuse you are using the throttle for the throttle. Jerking the throttle will result in a throttle response.

And actually right now, if you set up a reverse button, it will regen brake before going into reverse (applies current map in opposite direction)

1ofko6, 2022-07-09:
Well, the button for changing modes isnt a bad idea... but as we know, usually from baterry compartment on bike goes 
JULET  5 pin connector.  I have +5V,GND,ADC1,TX,RX .    
So thats the reason why I would like to test throttle gestures..   
Did I explain it well enough?

1ofko6, 2022-07-09:
I have no wire for mode change or for ADC2...

geofrancis, 2022-07-10:
what your suggesting won’t work, because when you put the throttle from zero max to try and set it is going to cause it to accelerate at maximum power.

1ofko6, 2022-07-11:
Nope, if you have safe start...
Of course i found out that there is feature like that..
Its called parking brake or hand brake..
Read something about it and than you will find out that it should be very easy to do and pretty much safe😃

closed 2022-07-17 by 1ofko6