vedderb/bldc #519: Incorrect and unphysical current values (I_q, I_in, I_motor)

flytrex-vadim, 2022-08-29:
Any ideas on how to approach investigating this further would be greatly appreciated.

### Setup
seeing this on both 4.10 and 6 hardware configs, FW versions as new as 5.03 and as old as 3.40.
Motors: TMotor MN series outrunners (4010, 5012, 601) and Faulhaber BP4 inrunners
Power: 6S Li battery
Control mode: speed control
Load: none

### Expected behavior
match between reported current values and ampere-meter readings. positive current and power values for drive quadrants, negative for generating quadrants. 

### Observed result
reported current is very different from measured, the discrepancy varies with RPM. For some motors (especially inrunners) the reported current is negative for wide ranges of speed. This causes all current limiting functionality to be nearly useless.

### What I tired so far

1. Calibrated phase shunt current measurement by comparing current commanded to mearure_res command with the ampere-meter reading of phase current. all looks good.
2. Running with encoder commutation (5047)  instead of sensorless commutation to rule out phase mismatch - no difference

### Plots
MN5012 motor - RPM sweep
reported current drops at high RPM
[image]

BP4 motor - RPM sweep
reported current is **negative**
[image]

TechAUmNu, 2022-09-08:
What esc are you using?

flytrex-vadim, 2022-09-09:
v4 hardware is a Maytech's MTVESC50, and v6 is our own board design based on VESC6 MK2 schematics.
It appears that both those designs have a current sense RC constant around 0.5us. 
I've now tried increasing the time constant to around 15us, in line with latest v6 mk3-5 designs, and it seems to have improved the situation greatly. Reported currents are now positive with most motors. 

The situation with Faulhaber BP4 is a bit more involved: its inductance is so low (electrical time constant  ~100us) that a much higher switching frequency would be needed. Since the CPU only seems to support  up to ~50KHz or so, I'm experimenting with adding series phase inductors to reduce ripple.