vedderb/bldc #520: lisp can commands missing: conf-set

1zun4, 2022-08-29:
Hello, so the following issue:

I got many reports of people who want to use the M365 Dash Script with Dual VESCs that is no problem until we want to limit speed for both VESCs.
```lisp
(conf-set 'max-speed (/ speed-sport 3.6))
(conf-set 'l-watt-max watt-sport)
```

Now we have only limited the speed of one VESC, but how do we sync it to the second VESC. Please add this feature or let me know if there is another way to sync two VESCs.

vedderb, 2022-12-02:
I thought about this for a while and came up with this solution:
https://github.com/vedderb/bldc/tree/master/lispBM#can-cmd
That way you can run any command on any can-device.

closed 2022-12-02 by vedderb

1zun4, 2022-12-03:
Tested and works flawless! Thank you.