vedderb/bldc #523: Add support for servo pin as GPIO

AkbarRamzan, 2022-08-31:
The number of pins you can use for GPIO is limited currently to:

'pin-rx     ; RX-pin on the COMM-port
'pin-tx     ; TX-pin on the COMM-port
'pin-swdio  ; IO-pin on the SWD-port
'pin-swclk  ; CLK-pin on the SWD-port
'pin-hall1  ; Sensor port hall1
'pin-hall2  ; Sensor port hall2
'pin-hall3  ; Sensor port hall3
'pin-adc1   ; ADC1-pin on COMM-port
'pin-adc2   ; ADC2-pin on COMM-port

It would be nice if this list was exapanded to also include things like the servo pin and any other pins might be available.

Mitchlol, 2022-09-10:
+1 for this, there are bunch of unused pins that are exposed on the lucas's controllers as well. 

LittleFOCer: PB12 PC5 PC14 PC15
CheapFocer2: PA15 PB2 PB3 PB4 PB12 PC5 PC13 PC14 PC15 PD2

vedderb, 2022-12-02:
Done a few commits back.

closed 2022-12-02 by vedderb