vedderb/bldc #525: Feature request: support for one wire PAS

AkbarRamzan, 2022-09-01:
Currently only quadrature sensors are supported but many (cheaper) PAS sensors use only a single wire.
There are 2 signal types for one wire PAS:
One pulse every time a magnet passes when pedalling forward, no signal when pedalling backwards
One pulse of a certain width when pedalling forwards, a different width when pedalling backwards. Backwards pedalling can be used for regen.
Ideally the servo pin should be the default as uart could be used for Bluetooth / display but ebikes are unlikely to use servo.

AkbarRamzan, 2022-09-01:
[image]

AkbarRamzan, 2022-09-06:
https://github.com/vedderb/bldc/pull/243#issuecomment-809470359
some code for PAS that may be useful.

AkbarRamzan, 2023-01-03:
[image]
I captured the signal generated when pedalling forwards using lisp at increasing RPM
[image]
It appears back-pedalling has <50% duty cycle so you can work out pedal direction that way. Back-pedalling for regen would be nice.

AkbarRamzan, 2023-03-26:
I'm adding support through lisp/ QML and packages. Stay tuned

closed 2023-03-26 by AkbarRamzan

ReFeXe, 2023-04-21:
How did you manage to implement this?
Have varied width pas, when using code https://github.com/vedderb/bldc/pull/243#issuecomment-809470359 
I get triggered both ways

ReFeXe, 2023-04-22:
> I'm adding support through lisp/ QML and packages. Stay tuned

I can't figure out how to adapt the above code

ReFeXe, 2023-04-22:
I did it, though.
https://github.com/ReFeXe/bldclcd3/blob/release_6_02/applications/app_pas.c