vedderb/bldc #527: Safety critical: throttle runaway event/ cruise control

AkbarRamzan, 2022-09-06:
It is possible for cruise control to become enabled randomly. I am not sure what’s causing it but I believe it may be related to the uart app. There needs to be a way to disable cruise control interpretation from the app!
Please add more details if you know about the issue. For now disabling the uart app seems to work.

vedderb, 2022-09-23:
I have not seen this myself. How do I reproduce this problem?

AkbarRamzan, 2022-09-23:
It's hard to reproduce but I think this is what is happening:
I have UART display connected (not supported by VESC)
The display sends packets which just so happens to correlate to enable cruise control
wheel therefore stays at constant speed until brake is pressed
In the UART app, I propose a way to disable cruise control setting

In this scenario it was my mistake, no need to enable uart app when using uart in lisp but other people have also been having this issue.

nitrousnrg, 2022-09-23:
Yes, it was your mistake at hardware level, something that an end user should normally not have access to.

Having said so, I did the same mistake a few years ago in a high power ebike so having the option to simply disable cruise control would be nice.

AkbarRamzan, 2022-09-27:
This can actually happen when there is nothing connected to UART port, as long as the UART app is enabled. Maybe from noise? Idk

AkbarRamzan, 2022-09-27:
Afzal said: "I think when uart app is enabled, servo will act as cruise control button (might depend on the board), and probably the signal on servo pin is causing the issue"

nitrousnrg, 2022-09-28:
Issue fixed in 35d2dfcff2ee1dbcb0ea23c0e52cf32a686317fd
You can disable cruise control now.

closed 2022-12-02 by vedderb