vedderb/bldc #528: Safety feature request: emergency stop app

AkbarRamzan, 2022-09-06:
It would be nice if there was an app that triggered an emergency stop if a given pin state becomes high or low. There should be support to select internal pull-up or pull down for the input. On ebikes, this would commonly be an ignition key switch or a magnetic clip that goes from your handlebar to your clothing so if you fall off, it detaches and triggers an emergency stop. This can be different to using ADC2 brake as you may want the vehicle to freewheel to a stop rather than regen brake.

vedderb, 2022-09-23:
There is already a kill switch setting in the app config where you can use the ppm or adc pin as a kill switch. Is this what you are looking for?

AkbarRamzan, 2022-09-23:
Yes actually that's exactly it, thanks

closed 2022-09-23 by AkbarRamzan