vedderb/bldc #529: Power limit should include time

AkbarRamzan, 2022-09-06:
U.K. and EU law have limitations on power but this limit is based on an average over 30 minutes. There’s no peak power limit.
A user should be able to specify 30 minutes when setting a power limit. 
Also the power limit is based on output, not input power so it would be nice if there was an option to try and ignore thermal losses by estimating efficiency.

vedderb, 2022-09-23:
This sounds very arbitrary and does not make much sense to me. What exactly does 30 minutes mean here?

AkbarRamzan, 2022-09-23:
Yes it is a very strange law but for electric vehicles the power must be limited in such a way that the mean average power output over any 30 minute window does not exceed the lawful limit. This allows an 250w ebike for example to peak at 1kW+ for steep hills. It also includes motorcycles such as the zero S 11kW (125cc category ) peak at 50kW and reach speeds of 90mph legally.

I am not sure on a good implementation but limiting instantaneous power is very limiting.

AkbarRamzan, 2022-09-23:
UNECE regulation No 85 is the standard
and the ESC is referred to as "Speed variator and control device"

kalvdans, 2022-09-23:
So I can ride for 10 minutes in 750W and then stop for 20 minutes before the VESC allow me to move again?

AkbarRamzan, 2022-09-23:
Yes in theory but practically you would only ride for 750W for a short burst of say 5 minutes and then run at say 200W for 25 minutes so that the average is under compliance. It’s a silly law but it’s just how it is. We need a good algorithm so that you don’t have to stop, it just throttles back

don’t forget that it’s output power so drawing 750W from the battery with 250W thermal loss only counts as 500W.

AkbarRamzan, 2022-09-23:
Regen is also a grey area, does it count as negative power?

geofrancis, 2022-09-23:
just leave it set for 250w/15mph and use the profiles on the app to temporally set a higher power level. when you turn the bike off and on its back to 250w. since the temporary setting is only held in ram.

kalvdans, 2022-09-23:
Haha! It would make more sense from a safety point of view to limit the total kinetic energy of vehicle + driver. Where do I send suggestions for law changes in EU now again?

geofrancis, 2022-09-23:
> Regen is also a grey area, does it count as negative power?

thats just facetious.

AkbarRamzan, 2022-09-23:
But you said 125Wh so I think +150Wh drive -25Wh regen = 125Wh is fine

AkbarRamzan, 2022-09-23:
Changing profiles would be illegal as you can exceed the power limits. Mostly a problem if you intend to sell bikes

geofrancis, 2022-09-23:
> But you said 125Wh so I think +150Wh drive -25Wh regen = 125Wh is fine

I said you can calculate it as 125wh in a 30 minute window , what part of 250w over 30 minutes are you unable to understand its a very basic concept that you somehow thing the back brake is involved with?

AkbarRamzan, 2022-09-23:
Because regen is the motor outputting a negative power so I think it should be included

geofrancis, 2022-09-23:
good for you, but legally what you think is irrelevant.

AkbarRamzan, 2022-09-23:
No need to be rude. I don’t see anything in the regulation that states it’s disallowed and it seems perfectly reasonable to include

AkbarRamzan, 2022-09-23:
Idea:
User specifies min, nominal and max power limit as well as a period
This could be for example min: 150W, nominal: 250W max: 1kW, period 30 minutes
VESC keeps track of past 30 minute energy output from motor and sets power limit to not break the min, nominal and max conditions
The 750W for 10 minute scenario cannot happen because it breaks the minimum 150W rule once that 10 minute is up since there's no energy budget left. Instead with this implementation, the vesc will start throttling back to ensure there is always enough energy to sustain 150W output but still give you a high peak power when required such as setting off uphill.

mckirkus, 2022-12-31:
May be better to just display kw used over the last 30 minutes to the end user.  So if pulled over you can show police that you've used less than the limit.  Limiting output based on this would be the equivalent of horrible file copy time estimates on windows, but deadlier.

AkbarRamzan, 2022-12-31:
I disagree and what you said isn’t legally sufficient anyway.
Other road legal motor controllers implement this already with no issue. If the bike was to be type tested, they will test the 30 minute nominal power output.

jaykup26, 2022-12-31:
Setting a cumulative 250w limit over a specified period of time could probably be programmed as a LISP extension

Would need to know:

- The exact language of the relevant sections of the law, to be able to determine requirements
- Determine how the algorithm should work.
  - How will you know output power without measuring it on a dyno?  Efficiency changes, sometimes significantly, with load and RPM.  Does estimating it work with the language of the current law?
  - Any speed limits?
- Should it be built into the tool or as a LISP script, loaded for those who need it?
- Does it need to be displayed anywhere?  How would you prove to an officer that you have been under the limit?
- Does a VESC based ebike need to be tested and certified?  How will this get implemented?  Is it tied to the firmware or hardware?  Is there a cost?  Who will pay for it?
- What if you only ride for 15 minutes?  Are there no limits?
- Is the ability to change profiles illegal, or the act of actually changing profiles?
- How does regen factor into this?
- How many datapoints does it need to average this to be legal?  One every second or every ms?

I think the ground work would need to be done first before any programming attempt can be made.  Basically, write a design spec, and maybe someone will be able to code it for you.

AkbarRamzan, 2022-12-31:
Yes, there may be a maximum speed limit. The power output needs to be reduced to 0W above this limit. Some manufacturers use virtual electronic freewheeling which is where the motor outputs a small power to overcome hysteresis losses of the motor itself. This is a power which is so small it is unable to propel the vehicle on flat ground but for a pedal assisted ebike makes it easier to pedal when the speed limit of 25kmh is reached.

Yes, estimates can be used. The technical standard actually talks about "the best estimate of the manufacturer" BUT this is then tested as part of the type testing procedure and must be within the allowable tolerances for it to pass. A poor estimate is not a problem as you can change your estimate and get it retested until it passes.

I think a lisp script could work though I am not sure how difficult it would be to implement it as a core feature.

Power consumption does not need to be displayed anywhere, the type approval documentation / markings etc would be sufficient.

If you only ride for 15 minutes, you must still abide by the average power limit. I guess a 250W ebike could therefore output 500W for 15 minutes and shut off but this would be an unrealistic scenario as you then couldn't ride with power for 15 minutes.

I'll come back with further answers.

vedderb, 2023-01-04:
> I think the ground work would need to be done first before any programming attempt can be made. Basically, write a design spec, and maybe someone will be able to code it for you.

100% agree on this. As it is now making anything that tries to approximate what has been suggested in this issue will be a lot if work that results in a feature that behaves like garbage and probably does not change anything legally.

vedderb, 2023-01-04:
Further, if the law allows a 500w ebike to run at 5kw if you only plan to take a quick 6 minute trip to the store the law is completely silly and needs to be updated.

AkbarRamzan, 2023-01-04:
Bosch uses this law to allow their 250W certified motors to peak at over 1kW and pull insane torque figures for climbing steep hills. I am not sure what kind of algorithm they use.

I have my bike type approved as a special category low power moped (makes throttle legal) but they didn't actually test the motor power, I had to self-certify that it was power limited.

vedderb, 2023-01-04:
Still, I don't feel like putting effort into making a technically convoluted and useless algorithm to comply with a silly law that will change anyway if the lawmakers gain some sanity. I would only implement something that has a reason to exist because it adds some real-world value independent of working around legal stuff written in a silly way.

That being said, I have my doubts that this actually is what the law means and I think there will be major differences in different countries.

vedderb, 2023-01-04:
If you want to mess with that and comply with your local laws when making your own type certificate you have all the tools you need in the scripting.

AkbarRamzan, 2023-01-04:
I can assure you it covers most of the EU and the U.K. It’s the same for motorcycles, there’s an 11kW class but zero motorcycles have a bike which peaks at 50kW and can do 90mph

I’ll stick with lisp script for now.

closed 2023-01-04 by AkbarRamzan