vedderb/bldc #553: App Not RUN after reboot or power cycle

trisuliswanto, 2022-11-17:
App Will Not RUN after reboot or power cycle, the reason is because in the main startup program app_set_configuration there is a check condition that will check if app has changed, and if there is no app changed the APP will not start
bool app_changed = appconf.app_to_use != conf->app_to_use;

trisuliswanto, 2022-11-17:
I use ADC APP and it won't run, but as soon as I change another app it begin to run..

vedderb, 2022-11-17:
I tried it with adc and adc + uart now, both start correctly after a power cycle. Do you perhaps have some script running that disables the adc-app?

trisuliswanto, 2022-11-17:
oh sorry, it's my bad, I forget to remove #define HW_HAS_PERMANENT_NRF in my hwconf, now it start correctly.

closed 2022-11-17 by trisuliswanto