vedderb/bldc #557: Compiling in Windows (PowerShell) error

Talabo, 2022-11-21:
Compilation of the master-branch in Windows 11 (Windows PowerShell) seem to break and not work "straight from the shell". With the setup instructions from the masterbranch readme file.

This is the following result I receive:

> PS C:\Users\me\Downloads\bldc> make 100_250
> "********* BUILD: 100_250 **********"
> 
> 
>     Directory: C:\Users\me\Downloads\bldc\build
> 
> 
> Mode                 LastWriteTime         Length Name
> ----                 -------------         ------ ----
> d-----        18.11.2022     11:04                100_250
> 
> 
> make[1]: Entering directory 'C:/Users/me/Downloads/bldc'
> The syntax of the command is incorrect.
> Compiling crt0_v7m.s
> ChibiOS_3.0.5/os/common/ports/ARMCMx/compilers/GCC/crt0_v7m.s:1:0: fatal error: opening dependency file build/100_250/.dep/crt0_v7m.o.d: No such file or directory
>  /*
> 
> compilation terminated.
> make[1]: *** [ChibiOS_3.0.5/os/common/ports/ARMCMx/compilers/GCC/rules.mk:235: C:/Users/me/Downloads/bldc/build/100_250/obj/crt0_v7m.o] Error 1
> make[1]: Leaving directory 'C:/Users/me/Downloads/bldc'
> make: *** [Makefile:212: fw_100_250_vescfw] Error 2
> PS C:\Users\me\Downloads\bldc>

If i run `make all_ut` I get the following:

> process_begin: CreateProcess(NULL, g++ --version, ...) failed.
> Makefile:18: pipe: No error
> Makefile:24: Unknown compiler
> make[1]: *** No rule to make target 'C:/Users/me/Downloads/bldc/utils_math.c', needed by 'C:/Users/me/Downloads/bldc/build/unit_tests/utils_math/utils_math.o'.  Stop.
> make: *** [Makefile:312: ut_utils_math_elf] Error 2

Teslafly, 2022-11-21:
It's missing the `.dep` folder. If you manually make that as a workaround it should compile fine.

This does need to be fixed though.

Teslafly, 2022-11-21:
I think that the issue is this line:
`$(shell cmd /C if not exist "$(DEPPATH)" mkdir "$(DEPPATH)")`
it gives the error `The syntax of the command is incorrect.`

Shortening it to `$(shell mkdir "$(DEPPATH)")` works and compiles after a clean. but complains the directory already exists.
`A subdirectory or file build/60/.dep already exists.`

I can give a try at fixing it in a day or two. Some cursory poking at it hasn't yielded any true quick fixes.

Talabo, 2022-11-21:
> I think that the issue is this line: `$(shell cmd /C if not exist "$(DEPPATH)" mkdir "$(DEPPATH)")` it gives the error `The syntax of the command is incorrect.`
>
> Shortening it to `$(shell mkdir [...]

> This is the commit that most likely caused this: [1dbb144](https://github.com/vedderb/bldc/commit/1dbb1442028258b7b671a19f2e1da7637acf4e39) I have no way to test this on windows, so it would be good if some windows-person can have a look. Reverting the commit is not an option as the dependencies were not working properly before that.

I'll look into both of this as well during the week. Thanks for the tips!

lipk888, 2022-11-27:
Hi Team:
same issue for me.
I follow Teslafly's comments and try to manually created ".dep" fold in target build fold and it worked fine.
it can be a alert solution. Thanks for help 

[image]

Jfriesen222, 2022-12-01:
I think we can close this issue unless anyone is still seeing issues

Teslafly, 2022-12-02:
Yes. This seems to be fixed/closed by https://github.com/vedderb/bldc/pull/562
Just successfully tried in on my Windows 11 system.

Tomblarom, 2023-11-13:
I encountered a similar issue and I had it compiling before. Had issues like `Error 2` and crt0_v7m.o.d: No such file or directory`

Turns out I forgot to run `make arm_sdk_install` after cloning.. Hope this helps somebody.