vedderb/bldc #559: Access to config parameters / mc_interface functions

surfdado, 2022-11-22:
Hi, here's a hopefully complete list of all the config params and runtime data/functions we'd like to access from the balance app package:

1) IMU Access:
imu_conf.mahony_kp (RW)
imu_conf.rot_roll, rot_pitch, rot_yaw (RW)
call to imu_init (even after initialization step)

2) Motor Config Access:
CFG_PARAM_temp_fet_start
CFG_PARAM_temp_motor_start
CFG_PARAM_max_duty
erpm_min/max and current min/max seem to be there already

Additionally I'd also like a way to set erpm min/max from the C-code

3) Runtime values: The below already seem to exist, so we should be good - just not sure if set_odometer also writes to flash

// get temperatures to perform tiltback to alert user
mc_interface_temp_fet_filtered()
mc_interface_temp_motor_filtered()

// get current odometer, update/store odometer
mc_interface_get_odometer() + write odometer

vedderb, 2022-11-22:
I can add these settings.

Why do you need imu_init? If I add it I also need to expose the imu-config which puts a large constraint in not breaking compatibility in future upgrades.

vedderb, 2022-11-22:
set_odometer will not write to flash, but I can add conf_general_store_backup_data which does. It has to be used with care though as the motor will stop for a few seconds when using it. You also don't want to write to flash too often to not wear it out. As it is now it tries to store it to flash when the power switch is used.

vedderb, 2022-11-22:
All done, except imu_init:
https://github.com/vedderb/bldc/commit/297e3ad43438a95056509b9184aba4bb6c0487c4
If you change the imu-parameters and an init is needed that will be done automatically

surfdado, 2022-11-22:
thank you benjamin, this is excellent - to see why I wanted imu_init, here's a little video (watch 20 seconds from the moment linked) https://youtu.be/DhMkGlqP_XE?t=1151

but your approach should work too instead - so when I change the IMU rotation values, imu_init will be called automatically?
However, when I change mahony_kp, I'm hoping imu_init will not be called, right? Because that is a setting we want to change while still riding

about the odometer/flash, yes I am well aware of the constraints so the current policy in my balance app is to write it any time the user takes a brake of more than 10 seconds AND the odometer delta is > 100m - even my Tramp 6_75 board doesn't have the fancy power button that would allow odometer writing

surfdado, 2022-11-22:
Oh, interesting - I never tried rotating without re-init, this is even better then - thank you!

closed 2022-12-02 by vedderb