vedderb/bldc #572: Lisp crashes on CPU overload (most likely)

1zun4, 2022-12-03:
For example on the [M365 VESC Dash](https://github.com/m365fw/vesc_m365_dash), we have following issue:
When the Zero Vector Frequency is too high, it will cause the Lisp script too crash, this means no input will be read from the dashboard anymore and it stays fixed on the last input until the timeout is reached. When this happens, the VESC has to be restarted to restart the script again.
_(Please add "Restart VESC Package" button)_

So far I was not able to further debug this issue without any errors showing up.
The only approach I can do is trying to further optimize the script by writing it natively instead of the Lisp language.

vedderb, 2022-12-03:
I had a quick look at the code. Do you think it could be because the length you read gets corrupt or a byte gets skipped so that you try to read more bytes than there is space in the buffer?

[image]

uart-read-bytes will give you eval_error if you ask it to read more bytes than the buffer has space for, which would stop that thread.

Also, how much cpu is the code using? It would be fairly easy to make a C function that gives you an extension to calculate the crc for example.

1zun4, 2022-12-05:
> uart-read-bytes will give you eval_error if you ask it to read more bytes than the buffer has space for, which would stop that thread.

Thank you. Good catch. I'll fix it and try again.
I report back with more details soon.

closed 2023-01-18 by 1zun4