vedderb/bldc #584: Balance App Refinement (suggestion)

Aquaharmonics, 2022-12-21:
I started looking at the balance app code and have two suggestions to make, assuming my understanding is correct. Please let me know if it is not. 

First:As we start pushing these motors and systems harder, it seems inevitable that the motors will get hot and the thermal limits will enable. 

If my understanding is correct, the app uses duty cycle as a limit on when to pushback to cause the rider to slow down.  It seems like it will be desirable to have the duty cycle set-point not be fixed, but rather change proportionally with motor temperature from the maximum duty cycle desired to zero duty cycle as the thermal limit threshold is passed up to the maximum temperature limit where output current is zero, otherwise the rider could experience situations where the speed can be achieved per the duty cycle set point but not have much current/torque margin for safe operation. 

Maybe it is already doing something like this but I have missed it. 

Second: Similarly, instead of having only a low voltage set point , adjust the duty cycle set point proportionally with the bus/battery voltage so that there is smooth transition in operation up to the low voltage limit, and also so if the battery pack physics or BMS start to limit total power output of the battery pack, a condition where the rider can achieve a certain speed but then not have much current / torque margin available can't occur. 

Note on the last point , if one has a sufficiently large battery capacity this could be prevented or if the duty cycle set point is set low it could also be prevented to some degree. 

Let me know if I'm misunderstanding the code function but I hope to test the hardware out in these cases to improve my understanding as well.

Aquaharmonics, 2022-12-27:
`/*
	Copyright 2019 Mitch Lustig
	This file is part of the VESC firmware.
	The VESC firmware is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.
    The VESC firmware is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.
    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
    */

#include "conf_general.h"

#include "ch.h" // ChibiOS
#include "hal.h" // ChibiOS HAL
#include "mc_interface.h" // Motor control functions
#include "hw.h" // Pin mapping on this hardware
#include "timeout.h" // To reset the timeout
#include "commands.h"
#include "imu/imu.h"
#include "imu/ahrs.h"
#include "utils.h"
#include "datatypes.h"
#include "comm_can.h"
#include "buzzer.h"

#include <math.h>

// Can
#define MAX_CAN_AGE 0.1

// Data type
typedef enum {
	STARTUP = 0,
	RUNNING,
	RUNNING_TILTBACK_DUTY,
	RUNNING_TILTBACK_HIGH_VOLTAGE,
	RUNNING_TILTBACK_LOW_VOLTAGE,
	RUNNING_TILTBACK_CONSTANT,
	FAULT_ANGLE_PITCH,
	FAULT_ANGLE_ROLL,
	FAULT_SWITCH_HALF,
	FAULT_SWITCH_FULL,
	FAULT_DUTY,
	FAULT_STARTUP
} BalanceState;

typedef enum {
	CENTERING = 0,
	TILTBACK
} SetpointAdjustmentType;

typedef enum {
	OFF = 0,
	HALF,
	ON
} SwitchState;

typedef enum {
	RIDE_OFF = 0,
	RIDE_IDLE = 1,
	RIDE_FORWARD,
	RIDE_REVERSE,
	BRAKE_FORWARD,
	BRAKE_REVERSE
} RideState;
[... 655 lines cut ...]
					break;
				}
				// Disable output
				brake();
				break;
			case (FAULT_DUTY):
				new_ride_state = RIDE_OFF;
				// We need another fault to clear duty fault.
				// Otherwise duty fault will clear itself as soon as motor pauses, then motor will spool up again.
				// Rendering this fault useless.
				check_faults(true);
				// Disable output
				brake();
				break;
		}

		// Delay between loops
		chThdSleepMicroseconds((int)((1000.0 / balance_conf.hertz) * 1000.0));
	}
	// in case we leave this force the buzzer off (force=regardless of ongoing multi beeps)
	beep_off(true);

	// Disable output
	brake();
}`

surfdado, 2022-12-29:
Actually both of your concerns are already covered in the float package and the ATR version of the balance app. I suggest you take a closer look at those to gain further understanding.

Aquaharmonics, 2022-12-29:
Hi @surfdado , thanks I will look into that, really neat! It looks like these are not GitHub repo's?

surfdado, 2022-12-29:
Look for vedder's vesc_pkg repo. In there you will find the float package

Aquaharmonics, 2022-12-29:
Got it thanks @surfdado .

I am going to close this for now. After another discussion with @surfdado, since the duty cycle varies linearly-proportionally with the bus/battery voltage already, it is already compensating in the way I proposed in my second item. As for the first, varying the duty-cycle setpoint based on the MOSFET and Motor temperature limits, this seems like it would still be an issue in the VESC firmware, but it is already addressed in the ATR and float package of the balance app.

closed 2022-12-29 by Aquaharmonics