vedderb/bldc #587: Feature request: PWM support for lisp

AkbarRamzan, 2023-01-02:
Currently the only PWM feature is ioboard-set-pwm.
Can we also write PWM to regular GPIO?
Can we also add a function to read a PWM signal?
E.g. get-pwm-freq (pin, sample_period, min_width, max_width) where sample_period is the time window in which to count over in seconds and the width is optionally defined to only count pulses that are within this width value, also in seconds.

the use-case for this is ebike single wire PAS which creates a pulse of certain width when pedalling forward and a different width when back pedalling. 95% of ebikes use this PAS but vesc only supports quadrature (2 signal wires).

svenkvade, 2023-02-05:
What about the Input Capture (ICU) commands? Wouldn't that work?

AkbarRamzan, 2023-03-26:
Yep, that new feature covers this issue. Perfect!

closed 2023-03-26 by AkbarRamzan