vedderb/bldc #589: Feature request: Alternating Reversing sequence & Null=V0 space vector PWM

Rampagy, 2023-01-12:
I noticed on a lot of the manuals for the VESC6 boards mention that FOC only utilizes sinusoidal commutation.  I have done some work with NULL=V0 and alternating reversing sequence space vector pulse width modulation on [my Github](https://github.com/Rampagy/bldc-firmware) and would like to try and implement it here.  I tried looking through the code to see if I could find if was already implemented, but I was not able to.

If you're willing to provide an overview of the code architecture and where you think a good injection point for this would be, it would greatly help me in my endeavor.

https://youtu.be/5eQyoVMz1dY
https://youtu.be/hSXSY4LRizg
https://youtu.be/oHEVdXucSJs

These commutation methods provide ~15% more phase to phase voltage than sinusoidal commutation, and I think it would be a great addition to the project.

[image]

Rampagy, 2023-01-12:
I found [the code](https://github.com/vedderb/bldc/blob/master/motor/foc_math.c#L206-L338) and [the description](https://github.com/vedderb/bldc/pull/372#issuecomment-962499623) of the code.

If you have alpha and beta would it be possible to directly calculate the quadrature angle with `tanh(beta/alpha)` and then use a lookup table with the quadrature angle?

I'm thinking something like this (this function assumes complimentary switching):

https://github.com/Rampagy/bldc-firmware/blob/main/CommutationLookupTable.c#L4-L64

with the following lookup tables:

https://github.com/Rampagy/bldc-firmware/blob/main/include/CommutationLookupTable.ch

Let me know what your thoughts are.

Rampagy, 2023-01-13:
I have gone ahead and [analyzed the code](https://github.com/Rampagy/bldc-firmware/tree/main/tests/vedderb) and I believe that there might be a bug or an improper interpretation of alpha and beta.

This is a plot I got by compiling [foc_svm()](https://github.com/vedderb/bldc/blob/master/motor/foc_math.c#L206-L338) in gcc to generate the three waveforms.  What perplexes me is that the duty cycle goes above the PWMFullDutyCycle and below 0. 

It's not possible for the duty cycle to actually go above  PWMFullDutyCycle or below 0 in the actual duty cycle.  It seems to me there is either a bug in the foc_math() function or I am not [interpreting alpha and beta](https://github.com/Rampagy/bldc-firmware/blob/main/tests/vedderb/test_vedderb_foc.c#L164-L165) correctly in my analysis.
[image]

ElwinBoots, 2023-01-17:
> I have gone ahead and [analyzed](https://github.com/Rampagy/bldc-firmware/tree/main/tests/vedderb) the [foc_svm()](https://github.com/vedderb/bldc/blob/master/motor/foc_math.c#L206-L338) function and I believe that there might be a bug or an improper interpretation of alpha and beta in my analysis.
> 
> This is a plot I got by compiling [foc_svm()](https://github.com/vedderb/bldc/blob/master/motor/foc_math.c#L206-L338) in gcc to generate the three waveforms. What perplexes me is that the duty cycle goes above PWMFullDutyCycle and below 0, as that's not actually possible on the microcontroller. [image]
> 
> It seems to me there is either a bug in the foc_math() function or I am not [interpreting alpha and beta](https://github.com/Rampagy/bldc-firmware/blob/main/tests/vedderb/test_vedderb_foc.c#L164-L165) correctly in my analysis.
> 
> The only thing I can think of that might be wrong with my interpretation is if the max alpha and beta could be was 0.866. If that's the case my analysis should use:
> 
> ```
> beta = 0.866f*(float)sin( (double)(i*3.14159f/180.0f) );
> alpha = 0.866f*(float)cos( (double)(i*3.14159f/180.0f) );
> ```
That is right. The mod_alpha and mod_beta go up to sqrt(3)/2.