vedderb/bldc #591: WS2812 or WS2811 with timer 2

sshenker85, 2023-01-17:
I see the latest code supports timer 3 and timer 4 (https://github.com/vedderb/bldc/tree/master/lispBM/c_libs/examples/ws2812/code.c)

Is there any reason it is limited to these 2 timers, is there any reason this cant be extended to support timer 2 for certain hardware needs? 

Thanks in advance.

Teslafly, 2023-01-18:
the chibios systick uses timer 2 unfortunately. timer 2 and timer 5 are the only 32 bit timers in the stm32f4 fit for chibios.
```#define STM32_ST_USE_TIMER                  2``` (in mcuconf.h)
Timers 1 and 8 are used for motor pwm generation and adc control.

Timers 3 and 4 are supported because they are the 2 timers historically available on extra io pins for most hardwares. Either on the hall sensor port or the servo port. also they support dma which i think is required?

I do not believe that timer 5 is being used for anything if one of it's gpio's is accessible. but it needs to be enabled.

timer 13 (PA6) or 14 (PA7) may be able to be used as well, though they may not be able to access DMA? which i think is required for ws2812 control.

[image]

Basically, the hardware timers are tightly integrated with the way vesc works and playing around with them without being careful and understanding the larger system can quickly break things.

what reason do you want to use a different timer?

sshenker85, 2023-01-18:
Appreciate your quick response. 

Basically I have a hardware design where I am utilizing almost all of the available pins including timer3 and timer4. PB3/ Timer2 is the only remaining pin which I could use for the LEDs as I also noted it has DMA access.    

If you say that timer5 could potentially be used for ChibiOS control, would it be a matter of simply swapping 2 to 5 in the below to free up timer 2 for the leds?:
#define STM32_ST_USE_TIMER    5 

Thanks again for your help.

Teslafly, 2023-01-18:
I think that would work. Both timer 2 and 5 are 32 bit. Thy it and see if it works. 
Worst case the compiler yells at you or it doesn't boot.

Then getting the led package to work is the other thing.

sshenker85, 2023-01-18:
Thanks Ill give this a try ! Yeah the other part should not be too difficult.

closed 2023-01-18 by sshenker85