vedderb/bldc #592: how to decide the value of MCCONF_SIGNATURE?

htq398, 2023-01-30:
I found  that the wrong MCCONF_SIGNATURE will result to motor configuration update failed in the VESC TOOL. 
but, when I got the actual signature value through breakpoint, and changed the MCCONF_SIGNATURE to that value. The motor configuration update can be successed.

Teslafly, 2023-01-30:
The developer tab configuration editor can generate the firmware configuration .C and .H files.

This is probably something that should go under better documentation in general. 
https://github.com/vedderb/vesc_tool/issues/92

htq398, 2023-01-30:
> The developer tab configuration editor can generate the firmware configuration .C and .H files.
> 
> This is probably something that should go under better documentation in general. [vedderb/vesc_tool#92](https://github.com/vedderb/vesc_tool/issues/92)

Thanks! I got it!