vedderb/bldc #605: Flipsky hardware firmware naming inconsistency

tonymillion, 2023-03-17:
I noticed there were some new Flipsky hardwares added for 6.02, that had `FSESC_` prepended to the name, but there are some old firmwares added by @jaykup26 that dont have this prefix. It would be nice for consistency to have them all correctly prefixed.

I'm happy to do it, but thought @jaykup26 might want to (since they are his).

jaykup26, 2023-03-17:
Yeah the naming scheme is a bit of a mess.

Creating the header files for the original 75100 was done because Flipsky at the time stated that the firmware could not be updated.  The units were shipped with "75100" from Flipsky running 5.02

Then to fix this "firmware cannot be updated" problem, Flipsky modified the hardware and released a "version 2" to sort of match the 75_300_R2 hardware from Trampa instead of simply providing the header file.  It shipped with 75_300_R2.  The new custom firmware was named 75100_V2

Then they released the 75100 aluminum PCB... then the 75200... then the 75300...

Here are the current names as they show up in the VESC tool:

75100 (covers v1 boxy version)
75100_V2 (covers v2 boxy and aluminum pcb)
FSESC_75_200_ALU
FSESC75300

The issue with changing these names now, is that anyone running a controller with this firmware, will be unable to update it in the future unless they see the change on Github, go to the firmware archive on Github, manually download the new file, and manually load it using the custom file tab. That's a lot of extra steps that will surely cause a lot of confusion down the road.

Had we foreseen the full product line, or better yet, had Flipsky responded to countless inquiries to work together on this, I think the naming scheme could have been better thought out.

Maybe if there was a "translation" section added to the code where the VESC tool would know that "75100_V2" and "FSESC_75100_V2" were the same thing then it would allow for a seamless update.

Until then I think leaving it the way it is will cause the least headache for the end users.

tonymillion, 2023-03-17:
Yeah, I remember the esk8 forum posts ;) Thank you for that, by the way. I had just gotten one of those "small black box" 75100 with the non-alu blue PCB - those things would overheat by just looking at them sternly.

The reason I thought I'd open an issue, is I just went through the whole "track down the right firmware"thing again,  as I just got ahold of the alu pcb dual 75100 one and it is still being sold with the 75300 VESC firmware that will blow out if you don't disable phase filters. 

It took me a while to track down the firmware file to load onto the device and looking through the huge list of vescs with number based names, I had to double , triple check that I had grabbed the right one. Vesc tool didn't make it easy, by not having the ability to just view all firmware files it shipped with, and it wasn't initially easy to find, as the repo was split out at some point. 

That alone might be a good reason to unify the naming scheme, or at least I kind of see it like, old users who've previously flashed their vescs already know the score and if they are anything like me stay on top of development. New users who aren't as savvy might have a hard time even finding the correct firmware from the list and having to use the flash from file UI - so its not really seamless in any case (other than for the people who've already gone through it before).

I agree on the `A.K.A` thing for vesc tool for this kind of thing, although this wont help for the initial freshly bought bad firmware issue :(

On a side note the new Alu boards are absolutely amazing for heat dissipation... I never really rated Flipsky highly (who does), but they are cheap and were my first into to VESC (way back in the V4 days) but apart from the stupidity with the Firmware they've done a good job with the "FSESC75100 with Alu. PCB" models.