vedderb/bldc #606: Feature request: Lisp ADC detach more modes/ improvement

AkbarRamzan, 2023-03-26:
It would be nice if you could detach ADC1 but leave ADC2 attached. This is for safety reasons so you can manipulate throttle but keep the brake system direct and be reliable. I propose adding mode 4: ADC1 only and mode 5: ADC2 only

Currently it is:
(app-adc-detach mode state)
; Where
; mode : Select peripheral to detach from APP
;        - 0 All peripheral attached (no second argument)
;        - 1 ADC1/2
;        - 2 Buttons
;        - 3 ADC1/2 + Buttons
; state : Only when mode 1/2/3/4 - 1 detaches peripheral from APP, 0 attaches peripheral to APP

vedderb, 2023-03-29:
Here is an attempt
https://github.com/vedderb/bldc/commit/478a385bf78f0049cc898e2415ee44b9e544c076

I made it so that state controls the ADC, otherwise you don't get the combination of e.g. ADC1 only and buttons. Better would be to have a mode for the buttons as the first argument and a mode for the ADCs as the second argument, but that would stop old code from working so I opted for this solution that should be backwards compatible.

I will update the beta build later today.

AkbarRamzan, 2023-06-21:
Happy with this