vedderb/bldc #609: can-cmd: not limiting motor speed

1zun4, 2023-04-05:
Hello,

I am writing about an issue that I am facing with my [M365 Dash AWD](https://github.com/m365fw/vesc_m365_dash/blob/main/m365_dash_awd.lisp) script. Specifically, I am having trouble with the following code:

```lisp
(define drive-speed (/ 17 3.6))
(define drive-current 0.7)
(define drive-watts 500)

...

(defun configure-speed(speed watts current)
    (progn
        (conf-set 'max-speed speed)
        (conf-set 'l-watt-max watts)
        (conf-set 'l-current-max-scale current)
        (can-cmd slave-can (str-merge "(conf-set 'max-speed " (str-from-n speed) ")"))
        (can-cmd slave-can (str-merge "(conf-set 'l-watt-max " (str-from-n watts) ")"))
        (can-cmd slave-can (str-merge "(conf-set 'l-current-max-scale " (str-from-n current) ")"))
    )
)

...

(configure-speed drive-speed drive-watts drive-current)
```

It appears that this code may no longer be working correctly. I have received reports from users since version 6.02 that the slave motor is no longer being limited as intended. Unfortunately, I do not have an AWD setup to test this myself, but one of my users had tested the script when it was first implemented on version 6.00 BETA and it had worked well for them.

I am sorry that I cannot provide a detailed error message, but I am wondering if you might have changed something in the code that could have affected this function.

For context, the can-cmd function is used to execute a command on a CAN-device with a specified ID. The command is sent as a string and will be parsed and evaluated by the receiver. The can-cmd function has more overhead than other CAN-Commands, so if possible they should be used instead.

1zun4, 2023-04-05:
I have now done a small test with a user with following test script:
```lisp
(define slave-can 103)
(define drive-speed (/ 45 3.6))
(define drive-current 0.9)
(define drive-watts 750)

(defun configure-speed(speed watts current)
    (progn
        (conf-set 'max-speed speed)
        (conf-set 'l-watt-max watts)
        (conf-set 'l-current-max-scale current)
        (can-cmd slave-can (str-merge "(conf-set 'max-speed " (str-from-n speed) ")"))
        (can-cmd slave-can (str-merge "(conf-set 'l-watt-max " (str-from-n watts) ")"))
        (can-cmd slave-can (str-merge "(conf-set 'l-current-max-scale " (str-from-n current) ")"))
    )
)

(configure-speed drive-speed drive-watts drive-current)
```

and it appears it really does not function correctly.
[image]
[image]

1zun4, 2023-04-05:
There we're no changes on the can-cmd command since when it was added, so I might guess there is a configuration issue on the user-side?

vedderb, 2023-04-05:
I had a look and there was indeed a bug in can-cmd. The fix is pushed, should be in the next beta.

aka13-404, 2023-06-21:
For better availability via search engines, this problem is not exclusive to speed or current, no can-cmd commands work in 6.02, and the REPL replies with >t to any such command.

aka13-404, 2023-07-02:
This is strange, maybe it got broken again? For me it definately works. 

```
;;;Function to set config parameter to local and all specified vescs
(defun set-param (param value can_slave_id_list)
    (progn
        (conf-set param value)
        (loopforeach i can_slave_id_list
            (can-cmd i (str-merge "(conf-set " "'" (sym2str param) " " (str-from-n value) ")"))
        )
    )
)
```
This was working on the 6.05 from April, the syntaxis looks the same to yours.