vedderb/bldc #617: No timeout on ADC override in Lisp

1zun4, 2023-04-22:
I have noticed that there is no timeout on `ext_app_adc_override`
https://github.com/vedderb/bldc/blob/26e3a3cec938e635d17e20cb86eda164747e1a0b/lispBM/lispif_vesc_extensions.c#L1150
Also not on the `app_adc_adc1_override`
https://github.com/vedderb/bldc/blob/26e3a3cec938e635d17e20cb86eda164747e1a0b/applications/app_adc.c#L128

So I guess as soon the Lisp crashes or anything else happens you are going to lose any control about your vehicle.
I remember this happening to some of the people from SmartESC v2 (VESC fork for M365 controller) which used a script making use of the ADC override, but seems no one has reported it so far.

1zun4, 2023-05-12:
Please. This needs to be fixed. It is a high risk when using.

1zun4, 2023-06-16:
Seems to be fixed by https://github.com/vedderb/bldc/commit/7487e5fd44d1ea5be7219a842bfb608a73facc6b. Thank you! Leaving it open until I tested it myself.

vedderb, 2023-06-16:
Yeah, should be fixed there hopefully. I was going to update this issue but forgot about it.