vedderb/bldc #628: Add command to set target speed and max current - needed for EScooters

casainho, 2023-05-21:
My Fiido Q1S 12 inches wheel, seated micro scooter, with a hub direct drive motor that has very low torque and very high speed motor, runs at 160 kms/h(1)!! Is very dangerous to rotate the throttle by accident so I need to set the motor speed in RPM (COMM_SET_RPM) but also set the motor max current simultaneous. This is the way I think electric motorcycles work, even my Xiaomi M365, a popular stand up EScooter.

So, if I rotate a little the throttle, like 10%, I will send this new command COMM_SET_RPM_CURRENT to VESC, with 10% of EScooter max speed value as also 10% of max motor current value.

(1) https://www.youtube.com/watch?v=USxNeejr6yk

[image]

nitrousnrg, 2023-05-21:
I think that the speed is not surging because of the regen algorithm. When you do one-pedal driving it will naturally produce that behavior, whilst still controlling the motor torque.

You can try my implemetantion of it to see if thats what you're after. https://github.com/luna-cycle/vesc_pkg/tree/one_pedal_driving/one_pedal_driving

casainho, 2023-05-21:
@nitrousnrg I appreciate your answer. I must say that I did took as reference the EBike app for Bafang M600, thank you. I did the same for my Bafang M500, but using Pyhton firmware instead -- see here: https://opensourceebike.github.io/

Ah, that LISP is ugly for my eyes :)
So I am using an ESP32 board running Pyhton firmware and interfacing with VESC by UART.

I just did the pull request to add the command I was looking for. I tested in a few testing rides on the street and it seems to work well. Well, anyway, I replicated a CAN command that was already setting only the motor max current.

nitrousnrg, 2023-05-21:
> Ah, that LISP is ugly for my eyes :)
Oh for sure. Try installing that vesc package, it includes a graph that shows what it does.

casainho, 2023-05-22:
@nitrousnrg , I am being trying to understand your algorithm. Why do you relate battery current and motor current??

What I am struggling to understand is only this part: 1-Calculate max braking and accel current for the current speed

Thank you.

nitrousnrg, 2023-05-23:
That's a secondary feature for systems constrained by battery current. Imagine a motor that can flow 400Apk but with a battery rated for 50Adc. In those cases at very low speed your phase current gets limited by battery current.

At high speed your throttle range would be 0-400A, but the controller won't drive more than 50Amps or so. That feels bad, as 75% of the throttle range does nothing at high speed.
Kind of similar situation happens for the regen. The whole point of this is to have a smooth "one-pedal-driving" experience.

In any case, give the vesc package a try and see if you like it

casainho, 2023-05-23:
I got it, thanks!!

Gabrielerusso, 2023-05-24:
> That's a secondary feature for systems constrained by battery current. Imagine a motor that can flow 400Apk but with a battery rated for 50Adc. In those cases at very low speed your phase current [...]

simply use the same current fro battery and phase and you will get almost the same response on the complete rpm range of the motor

casainho, 2023-05-24:
@Gabrielerusso , thanks for the feedback. I am using VESC only to control the motor, and an external board to read the sensors, control things as the lights, the battery BMS switch, send the motor commands to VESC and control the display - and I do all this in ESP32 Pyhton firmware, including wireless communication with the display and battery BMS switch.

And I am playing with the VESC speed controller PID values to see if it is enough for my needs, and maybe you are correct.

nitrousnrg, 2023-05-24:
> simply use the same current fro battery and phase and you will get almost the same response on the complete rpm range of the motor.

If your throttle controls directly the battery current, at zero rpm you can do 400 phase amps with like 10% throttle, its impossible to control torque.
I think I could dynamically reference the the adc end to battery current max if that happens only when modulation is maxed out.

casainho probably saw the same behavior that I saw, you give little throttle to an scooter and it doesn't shoot to max rpm like the default vesc does, and he thought there is a speed limiter in place. I thought the same at that time, until I recently realized that the behavior is 100% related to the regen algorithm.

Gabrielerusso, 2023-05-25:
> If your throttle controls directly the battery current, at zero rpm you can do 400 phase amps with like 10% throttle, its impossible to control torque. I think I could dynamically reference the the [...]

Nope, vesc in FOC mode allows you to limit the phase current and battery current separately, so if you put the same current limit for battery and phase you will get an almost constant torque response. As said commercial vehicles also do this and use acceleration limits to prevent unwanted or dangerous situations.
Providing a rpm limit for the given throttle it's not usefull at all as it will give problem during real use with wheels on the ground.
You can write a small LISP script to do so my reading the rpm, deriving it to obtain radial acceleration and create a small acceleration controller that will limit the phase/battery current. I've an escooter myself running 240A battery and 700A phase and on the road by having all parameters properly tuned i've not got any problem (yes i'm using current control)