vedderb/bldc #630: No way to use the serial port when the motor is running

t812206236, 2023-05-23:
I wanted to export the imu data to esp32, but I found that when I accidentally touched the motor, the serial port data stopped sending. At this time, the vesc indicator turned from blue to green(driving mode). Is this a normal phenomenon? What can I do to use the serial port while the motor is running?

t812206236, 2023-05-23:
I use the balance app and the built-in imu. Is this the cause of the problem?

Gabrielerusso, 2023-05-25:
> I wanted to export the imu data to esp32, but I found that when I accidentally touched the motor, the serial port data stopped sending. At this time, the vesc indicator turned from blue to [...]

Yes you can use the serial port while running, for example the BLE module runs on the second "uart". A common communication problem while using the motor is due to Electromagnetic interference (EMI), noise and bad grounding, following some tips:

- Make sure the ESC and esp32 are properly grounded and that there is some sort of filtering cap on the ESP.
- Adapt UART Baud to wire lenght, wire kind and place them far away from phase wires.
- Possibly choose twisted pair if length is greater than 10/20cm, shielded cables if greater distances.
- USE CANBUS

I'm not familiar with the balance app and imu, you need to check if there is some exclusion rule in the use of one or another, but i don't think so as uart has it's own hardware interface in the ic.

t812206236, 2023-05-26:
@Gabrielerusso 

If it is due to emi and noise, I should be able to receive some scattered information instead of receiving nothing at all. 

Besides, there is another point that can be explained: when rotating the wheel, the motherboard enters the driving mode(balancing app), When the rotation stopped, I did not receive the serial port information until the indicator light returned to blue. 

So I firmly believe that balancing app was blocking my serial communications while it was running.

I will try to use the can protocol, the can bus module is on the way

t812206236, 2023-05-29:
I checked it again and it was electromagnetic interference, but strangely enough the Bluetooth was totally fine, so I turned it off first, it wasn't a firmware problem

closed 2023-05-29 by t812206236

HenryJia, 2025-04-30:
Hi,

I'm also experiencing this exact same issue. How did you get around the problem of electromagnetic interference? I tried switching to using a shielded multicore cable for UART and still experiencing the exact same problem.