vedderb/bldc #633: Mixed up config for FAULT_CODE_ABS_OVER_CURRENT

tallakt, 2023-05-30:
We have had issues with a motor tripping due to the FAULT_CODE_ABS_OVER_CURRENT fault. I was checking out the code and this caught my attention. The setting in the VESC tool states "true" as meaning to use the filtered value for current alarm FAULT_CODE_ABS_OVER_CURRENT. In the code this seems to be opposite. Can anyone more knowledgeable comment if this could be a bug?

https://github.com/vedderb/bldc/blob/1647009a23157e2bf9b7fea81f9156b337361a03/motor/gpdrive.c#L567

tallakt, 2023-05-31:
Hi. I had some more time to look at this today. 

It seems the system is working as expected. I discovered this through trial and error.

This being said, some setups I was looking at, the whole "slow abs max current" business seems to have little effect. Even with these settings, the VESC would trip at approximately the same threshold for max slow current;

- Absolute MAximum Current:  28 A
- Slow ABS Current Limit: true
- Zero Vector Frequency: 30 kHz
- Current Filter Constant: 0.05

The code of the filter looks like this

```c++
#define UTILS_LP_FAST(value, sample, filter_constant)	(value -= (filter_constant) * ((value) - (sample)))
```

maybe I could rewrite this to

```c++
#define UTILS_LP_FAST(value, sample, filter_constant)	(value = value * (1.0 - filter_constant) + filter_constant * sample)
```

We could express this also by sample time T and time constant of the filter tau

```c++
value = tau / (T + tau) * value + T / (T + tau) * sample 
```

So we see that: 

```c++
filter_constant = T / (T + tau)
``` 

```c++
tau = ((T - T * filter_constant) / filter_constant
``` 

Choosing the smallest filter constant possible 0.05 and a sampling time 1/30000 the longest time constant possible is 0.6 ms.

Could it be better if the filter constant could co all the way down to eg. 0.001 being 33 ms time constant? Or even smaller?

TechAUmNu, 2023-06-13:
Generally it is not recommended to use slow max overcurrent, it has led to a lot of destroyed hardware previously. Especially when it gets enabled to try and stop the tripping, rather than fixing the config problem that makes it trip in the first place.