vedderb/bldc #641: Safety Issue: Regen brake stuttering on my e-scooter

DerNoel3005, 2023-06-30:
Hello,

I'm experiencing problems with the regen braking. 

When attending the brake to a full stop it ends in a stuttering-like stop, making it dangerous for emergency braking. I tried every combination of settings possible (different observer gains, max current of 120A + - 120A, observers, pwm frequency and bldc, sensorless etc.), but still the problem remains.

My actual setup consists of a Flipsky 75100 with 6.02 fw installed built into the Ninebot Max, running @ 52V/14S with VSett 10+ 1400W 50H motor. I also experienced the same problem on the original ninebot max 30H motor and a generic 1200W 40H motor, making me worry about the overall safety of the e-scooter when in use of a vesc.

Here's the video (the chain-like sound at the beginning is just the bags attached to the scooter)

https://github.com/vedderb/bldc/assets/115374001/d3bfb398-92a1-43c8-a1f0-6eb90f283b35

As you can see when I brake (seeable from the minus current) it always ends unsmoothly, making the ride unsafe in case of sudden danger. I experienced many close-ups of vehicles due to this.

Therefore it would be great to find the fundamental issue of this behavior, as it's a safety concern for many that also drive a Ninebot Max like me. 

The ninebot max runs a LISP from 1zuna ([source](https://github.com/m365fw/vesc_m365_dash/blob/main/m365_dash.lisp)) and runs [these ](https://paste.ccbluex.net/c18763000)settings as motor parameter.

tonymillion, 2023-06-30:
As an alternative to that lisp code (which was originally for m365 dashboards) I wrote one specifically for ninebot dashboards: https://github.com/tonymillion/VescNinebotDash

Anyway, on to an answer: I'm guessing you don't have physical brakes in this setup?

Assuming that is the case, I'm guessing the problem you're seeing is because you're not commanding a hand break (or reverse current) so that the slower you go, the less effective electronic braking is becoming. This should make sense since the wheel is generating less self-braking current the slower it goes.

DerNoel3005, 2023-06-30:
Thanks for your answer Tony!

I'll look into the code written by you and yes, you were right, I don't have any "real" physical brakes asides from the drum brake, which in itself, is very weak without an e-brake.
Thus I'm forced to use the combination of front-drum brake + rear e-brake to come to a full stop. If one doesn't work right I won't brake right as you saw from the video.

Is the hand brake something I need to set up myself or is it already included in your code?