vedderb/bldc #642: Feature Request - Add Minimum Start Speed and Minimum Brake Speed for ADC

1zun4, 2023-07-26:
I would like to propose the addition of two essential features to the VESC motor controller firmware - "Minimum Start Speed" and "Minimum Brake Speed." These features aim to enhance safety and address some peculiar issues observed during motor controller operation.

1. Minimum Start Speed:
The inclusion of a minimum start speed parameter would significantly improve safety during motor activation. Currently, there is a risk of accidental throttle input, causing the motor to start abruptly and potentially posing a safety hazard. By setting a minimum start speed, users can ensure that the motor will not engage until reaching this predefined speed. This implementation will prevent unintended motor activation and reduce the risk of unexpected acceleration, thus making it safer to operate the vehicle.

2. Minimum Brake Speed:
Presently, the VESC firmware seems to engage braking even at zero speed, leading to unusual behavior upon activation. After being turned on and applying the brake, causes the motor to move backward or forward unexpectedly. Additionally, this behavior may result in strange noises. By introducing the minimum brake speed feature, users can specify a threshold speed below which the braking mechanism will not engage. This enhancement should prevent the erratic braking behavior.

1zun4, 2023-09-20:
Ok, we really need to figure something out that stops the motor brake from engaging at zero speed. Especially with sensorless but also with hall sensors it randomly drives backwards...

1zun4, 2023-09-20:
So, about the minimum brake speed, this is more urgent in my opinion, because it is quite a security risk. I am not sure if this is just a me issue, but it happens on multiple setups, but sometimes the electric brake will cause the motor to run backwards, when it lost "track?". This only happens after start-up or sometimes when rolling out.

1zun4, 2024-06-01:
Push

Finnn-glitch, 2024-06-01:
I need this too

Stoindl, 2024-06-01:
Yeah, i also need this changes to be done.

CRzx1337, 2024-06-01:
That would be pretty useful, I would like to have that too.

LuckyLuciano420, 2024-06-03:
Yea would be super useful for E-Scooter use,also for Safty reasons

meiketch, 2024-06-03:
Yes, I need this as well and I know a few other people who do as well. My motors sucks if starting from a stand still, having them at 1kph works like a charm.

Paddel89, 2024-06-06:
I would be exactly in favor of it too and would wish that if this were implemented it would be much safer and I would feel much more comfortable on the ninebot and Vsett. Thanks for that!

PlasmaCR, 2024-06-15:
I have been fiending for this feature to be added ever since I have first used a VESC. This would add an optional layer of safety to the system, which would be very greatly appreciated!

vedderb, 2024-06-16:
To me having the throttle do nothing until you reach a certain speed sounds unsafe and unpredictable. If the thing you are using is very powerful, but twisting the throttle does nothing it can come as a big surprise that it suddenly starts working once you give it a push. Even on the weak rental scooters the fact that I have to push them for the throttle to start working makes them feel unsafe to me; I want to get going and it does nothing when I give it throttle, then I give it a small push while giving it more throttle and it still does nothing, then I give it a bit stronger push and it suddenly goes - that throttle behavior is really my least favorite feature about them. On everything I use I very much prefer that the throttle works from the very start. There is still the argument that if you have a sensorless setup with a motor that is bad at VSS or HFI the throttle won't be smooth until you reach some speed, but in my opinion it is still better than having the seemingly broken throttle that then suddenly starts working when you move. There is also the problem that if there is something wrong with your throttle you won't find that out until you move, at which point that fault is more dangerous.

Regarding the brake, I don't see the problem with having it work the way it does now. On everything I have tried it on it is more or less 100% smooth all the way down to 0. The braking force actually approaches 0 as you reach 0 speed as it only does regenerative braking and never active braking. I have not once seen the motor run backwards when braking as you mentioned, but maybe in your case that is caused by current offsets and setting the regen current to 0 while having no load (or some other bad combination of settings).

So from my perspective your arguments are literally the opposite of what I experience. Before spending work on something and adding settings with the potential of confusing people with even more settings I would like to see a good argument for this feature, especially for the minimum braking speed as I haven't seen the braking behavior you describe. Regarding the braking behavior, it would be nice if you can open an issue about it and provide some steps to reproduce it so that I can see what is going on with your setup.

meiketch, 2024-06-16:
I don´t think its unsafe or unpredictable, in some countries this is als required by law. We are also talking about min speed of 1kph, to have the motors not stutter, when going offroad and you suddenly came to an halt with one motor. 

Also this wont confuse users much further as min speed should be easy to understand besides all the other complicated stuff to adjust. To not confuse users, just have this option deactivated by default.

Saying that you would have a harder time to find a faulty throttle isnt true either, you could either find it in the logs or just show the ADC values in the app.

The thing with the brakes seems like you missunderstood, its not that they are driving backwards while in motion, it happens when you hit the brake while stand still. I noticed this behavior on every VESC Scooter I encountered. This can be very dangerous, especially with higher power motors as they kick much harder. Someone could get injured when I show him how to control the scooter for example.

Im still looking forward to the feature, there are a lot of people who need this and dont agree with your security concerns.

LuckyLuciano420, 2024-06-16:
> To me having the throttle do nothing until you reach a certain speed sounds unsafe and unpredictable. If the thing you are using is very powerful, but twisting the throttle does nothing it can come [...]

LuckyLuciano420, 2024-06-16:
> To me having the throttle do nothing until you reach a certain speed sounds unsafe and unpredictable. If the thing you are using is very powerful, but twisting the throttle does nothing it can come [...]

It would be much safer, maybe not in your case, but in mine and for many others bc of that.

https://github.com/vedderb/bldc/assets/171622807/48c8d9ea-4f96-41ae-b1fb-a3b01fc2708b

vedderb, 2024-06-16:
> The thing with the brakes seems like you missunderstood, its not that they are driving backwards while in motion, it happens when you hit the brake while stand still. I noticed this behavior on [...]

Again, the brake issue you are talking about is something I have never seen, even when using the brake while standing still. Please open a separate issue about that and show me how I can reproduce what you are seeing. I have no idea how things are connected and configured on your scooter, so the video does not help.

> I don´t think its unsafe or unpredictable, in some countries this is als required by law. We are also talking about min speed of 1kph, to have the motors not stutter, when going offroad and you suddenly came to an halt with one motor.

You are not addressing any of the points I made. Also, if you use the argument "required by law" please provide some evidence that 1) that is true and 2) how adding that feature will make any difference in making your diy-scooter count as legal on public roads.

> Saying that you would have a harder time to find a faulty throttle isnt true either, you could either find it in the logs or just show the ADC values in the app.

I don't understand how you even can make this argument. When I build something and configure it I start by putting the wheel(s) it has above ground so that it cannot move, then I connect to it, configure things and test the throttle as I do that. Then when I see that the throttle behaves reasonable I put it on the ground and continue testing. If It requires that I test the thing in person or push the wheels with my fingers before I have any idea what the throttle does it is obviously more likely that I get hurt.

meiketch, 2024-06-16:
Sorry, but your arguments just seems to be bogus to me. You arent even adressing the issue we are up to, but yes about the brake I can open it as an issue, once my scooter is up and running again.

vedderb, 2024-06-16:
If you fail to address my points and call my arguments bogus my motivation to spend time on fixing your issues is zero.

vedderb, 2024-06-16:
Keep in mind that I think your arguments are bogus too, but I try to have a conversation and understand them at least.

geofrancis, 2024-06-16:
@vedderb you keep missing the point that it's a legal requirement in a lot of places regardless of your opinion.

vedderb, 2024-06-16:
> @vedderb you keep missing the point that it's a legal requirement in a lot of places regardless of your opinion.

I already responded the _exactly_ that point. Are you able to read?

geofrancis, 2024-06-16:
> > @vedderb you keep missing the point that it's a legal requirement in a lot of places regardless of your opinion.
> 
> I already responded to _exactly_ that point. Are you able to read?

your reply was basically "prove it"

vedderb, 2024-06-16:
In the past there have been many people asking me to implement complicated and weird features to meet legal requirements, especially when it comes to power and average power. Trying to do that would make things a lot more complicated and probably still meet those requirements. Therefore, if you can make a good argument for a feature other than your interpretation of legal requirements, you better provide a good reason. So yes, pleas prove it!

LuckyLuciano420, 2024-06-16:
In my video, I showed why it's a problem, especially with the brake, but also with the throttle if you accidentally hit it when it's standing still. Of course, on a Onewheel or skateboard, it's a different situation. However, for a bike or scooter, having this feature adds a nice layer of security and is a good option to have.

Finnn-glitch, 2024-06-16:
The D-Norm isnt technically a law, but youre still required to follow the rules written in it, when you want to make a street legal scooter. That wouldnt make the DIY scooter legal, but less obvious in case of a police inspection.

vedderb, 2024-06-16:
> In my video, I showed why it's a problem, especially with the brake, but also with the throttle if you accidentally hit it when it's standing still. Of course, on a Onewheel or skateboard, it's a [...]

This is getting funny... I also literally spelled out that your video does not help me and that I need more details as your braking issue is not happening to me on any setup I have.

Finnn-glitch, 2024-06-16:
The Issue happens to a lot of people. Directly connecting the Hall of the Brake lever to the ADC ports of the Controller. Everyone who uses E-Brake has this issue tbh.

vedderb, 2024-06-16:
Also, the argument "it is dangerous that the motor runs when pushing the throttle" is very strange. It does exactly what it is supposed to do. If you leave you car running in drive with the door open when kids are around you are doing something very irresponsible and adding a weird quirk to it does not solve that problem except in maybe a few very unusual cases.

vedderb, 2024-06-16:
> The Issue happens to a lot of people. Directly connecting the Hall of the Brake lever to the ADC ports of the Controller. Everyone who uses E-Brake has this issue tbh.

AGAIN, it does not happen to me and the only way I can do ANYTHING about it is if you give me detailed steps to reproduce it.

Finnn-glitch, 2024-06-16:
> Also, the argument "it is dangerous that the motor runs when pushing the throttle" is very strange. It does exactly what it is supposed to do. If you leave you car running in drive with the door [...]

So that means the Car shouldnt move when pressing the brake pedal xD

meiketch, 2024-06-16:
I dont see whats issue here right now. Is it hard to implement? Everyone in this thread says they this issue, just because you dont have it doesnt mean it doesnt exist. Just have the brake not working below 1kph, it doesnt works below that speed anyways.

LuckyLuciano420, 2024-06-16:
> This is getting funny... I also literally spelled out that your video does not help me and that I need more details as your braking issue is not happening to me on any setup I have.

Why is this funny? It's an issue many people have. To give you more details, the brake hall sensor is connected to the ADC port. It seems like when you press it, the motor is searching for the position. Regarding the throttle, it's just an additional layer of safety. What is so hard to understand?

vedderb, 2024-06-16:
> So that means the Car shouldnt move when pressing the brake pedal xD

This is getting exceptionally frustrating. I do NOT have the braking issue and I NEED steps to reproduce it to do ANYTHING.

vedderb, 2024-06-16:
This is going nowhere and the tone keeps getting works. Perfect way of ruining my Sunday afternoon, thank you!

Finnn-glitch, 2024-06-16:
Youre not forced to work on Sunday. Were just trying to express our points to you. The arguments seems like a biased opinion. " When i doesnt have the issue noone has this issue".

vedderb, 2024-06-16:
> I dont see whats issue here right now. Is it hard to implement? Everyone in this thread says they this issue, just because you dont have it doesnt mean it doesnt exist. Just have the brake not working below 1kph, it doesnt works below that speed anyways.

Why is this so hard to understand? Even if every single person in the world says this (which is NOT the case, there are literally thousands of other people not having it as far as I know) I cannot fix it if I cannot reproduce it. There are millions of different ways to combine motors, batteries and configuration parameters and there is no way I can test all of them. I have spent vast amounts of effort on trying everything for years and hunting down all issues I can find. If you see something I haven't seen I absolutely need all details about it.

vedderb, 2024-06-16:
Btw, this also highlights why I don't want to add even more complexity and parameters unless there is a good reason. It increases the potential for bugs in the interaction with other parameters and use cases.

LuckyLuciano420, 2024-06-16:
> Btw, this also highlights why I don't want to add even more complexity and parameters unless there is a good reason. It increases the potential for bugs in the interaction with other parameters and use cases.

Yeah, it's not a good reason to add extra security by implementing a minimum speed for brake and throttle, but having the motors play MP3 files is something the world needs.

At this point, I don't care if it gets added or not. Maybe it can be fixed with Lisp, at least.

LuckyLuciano420, 2024-06-16:
Also no wonder why ppl leave the Vesc Community...

Finnn-glitch, 2024-06-16:
True

Finnn-glitch, 2024-06-16:
Maybe you see it from the wrong side. Maybe see it from a Feature side which also fixes an annoying issue

vedderb, 2024-06-16:
Incredibly enough you are still completely ignore everything I write about the break issue, that I don't have it myself and that I need detailed steps to reproduce it to fix it. This is completely ridiculous.

> Yeah, it's not a good reason to add extra security by implementing a minimum speed for brake and throttle, but having the motors play MP3 files is something the world needs.

I'm not against adding features that make things better, such as the ability to play audio with the motor (which people have been asking about for years btw). A neat implementation of audio playback naturally has the consequence that you can do a, in my opinion, cool demo and play Mp3. The mp3-code is not in the firmware btw, it is in scripts and vesc tool and demonstrates the flexibility of the implementation.

My points about your feature are that, as I have painfully explained without you listening, that it makes things worse! It is hard for me to spend my limited precious time on adding complexity that makes things worse. Also, again, I'm willing to implement it if you give me GOOD reasons, but all you do is ignore 80 % of my arguments, call what I write bogus, downvote my comments and say other hurtful things. Please, by all means, leave the community and please don't contact me again!

closed 2024-06-16 by vedderb

Finnn-glitch, 2024-06-16:
> Incredibly enough you are still completely ignore everything I write about the break issue, that I don't have it myself and that I need detailed steps to reproduce it to fix it. This is completely [...]

Yeah wea all ignore your points sure..

vedderb, 2024-06-16:
> Yeah wea all ignore your points sure..

Here you just ignored this thing I wrote: "... please don't contact me again!"

Finnn-glitch, 2024-06-16:
Ooohh sooorrryy

tonymillion, 2024-06-16:
To add to an already very noisy thread...

I have seen the "(sometimes) sudden jerk forward when the brake is released" and I looked into the cause. It's happened on every Flipsky controller I've had and on a Spintend UBOX v2. 

During debugging I got as far as deciding it's some kind of electrical interference/crosstalk on the ADC1/ADC2 pins. It's very obvious when you activate RT updates on the app and look at the ADC levels, pulling the brake will pull the level high and you can clearly see some leakage into the ADC1 line.

What happens next when you let go of the brake lever is the voltage level on ADC2 drops below the level on ADC1 and thus  the controller (rightly so) commands a forward current.

(one way to avoid this is to slowly release the brake and the crosstalk voltage to the ADC1 line will slowly descend with it and not trigger a motion)

the way I fixed it was to use different resistors in my 5v->3v resistor network ... I also tried tying the signal line to ground with a high value resistor....

what I'm saying is, the problem was on the wiring from the throttle & brake and not in vesc

vedderb, 2024-06-16:
That is a good hint, thanks for the input! It is hard to tell if the others had the same issue, but it does not sound unlikely. It shows why it is so hard to debug there things and why all details matter.

@1zun4
I'm happy to discuss this feature again in another feature request. I closed this one because the conversation went too far in a bad direction.

If you open it again, can you first look into the braking problem separately and open an issue on how to reproduce it? On my setups using the brake while standing still does almost nothing, maybe a tiny click if you listen carefully and some modulation noise in the motor. If it actually moves the motor in any direction I think there is another problem or potentially a bug that I haven't discovered yet that needs fixing.

1zun4, 2024-06-16:
> If you open it again, can you first look into the braking problem separately and open an issue on how to reproduce it? On my setups using the brake while standing still does almost nothing, maybe a tiny click if you listen carefully and some modulation noise in the motor. If it actually moves the motor in any direction I think there is another problem (such as the analog crosstalk described by @tonymillion) or potentially a bug that we haven't discovered yet that needs fixing.

I understand that applying minimum break speed is just a plaster on the wound. I will try to collect some data on this brake behaviour issue in order to get this issue resolved.

> To me having the throttle do nothing until you reach a certain speed sounds unsafe and unpredictable.

However, I am disappointed with your decision on motor start speed, if you think it is not worth implementing for VESC, can the use of `(app-disable-output -1)` to temporarily disable motor operation be used for this via Lisp?
I am currently using this function to make the display lock and shutdown work as expected while allowing the VESC to continue running. But I have not yet tried to implement a starting speed with it.

> Youre not forced to work on Sunday.

As for not responding on Sundays, being an open source maintainer is not that easy. You do not work in a company with fixed hours, where you can just not care about the product and users and keep the ticket open until next Monday. As an open source maintainer, you have dedicated yourself to a project out of personal interest, which means that you are willing to take care of it even on Sundays.

> Perfect way of ruining my Sunday afternoon, thank you!

But from the user's point of view, putting the unhappiness on the user may not be the right way. I really hate discussing issues with users myself, often without any information, but a: pls fix now, right now - does not apply to everyone here, but a video is simply not enough to diagnose such issues. It's like showing a pop-up dialog that says "An error occured".

\
Well, that is my contribution to this thread. Good Sunday afternoon, and if you feel the need to respond, don't.

vedderb, 2024-06-16:
I'm happy to respond and make progress any day when I'm available to do so. The problem is that I really put in effort to explain my view so that people can take that into account and make progress, but the response I get is a lot of negativity and most of my points getting completely ignored. The fact that I disagree with users here does not mean that I'm trying to pick a fight or win an argument for the sake of doing so, it means that I genuinely look for a good reason to implement something that sounds like a bad feature to me given my  background. If you can address my points and make a good argument for it I'm happy to do it.

Another big point about this request is that the braking issue seems to be the most critical here, and that should not be an issue in the first place. Rather than making a clunky workaround I would like to address the root cause. For doing so I need information and cooperation, not negativity, downvotes and useless comments such as "everyone has this issue, please fix".

I know that you are not responsible for statements others make, so feel no obligation to defend their comments. I'm happy you took time to make this feature request and explain your situation and reasoning. The reason I closed it is mainly that I think the useless conversation that happened today will get in the way, but also that I think that the braking issue should be addressed separately.

When it comes to the minimum speed for throttle feature, what I'm looking for is simply an explanation that makes me feel "ah, now it makes sense!". The throttle not working when getting started on scooters is something that always annoyed me greatly and every time I used them, especially when starting uphill at a traffic light, I asked myself what have they been thinking when they implemented this behavior that gets in the way all the time and does not help in any practical or theoretical situation I have ever encountered.