vedderb/bldc #655: SAFETY RISK: lisp app-disable-output causes ADC app to be locked to a certain speed

1zun4, 2023-09-20:
So, I experienced the following and I still have to confirm this correctly, but from what I could tell it seems  `app-disable-output` will block the ADC code and cause the motor to continue running at the last throttle value.

https://github.com/vedderb/bldc/blob/4e0288d37f203b58acacd735df81ce397319102f/applications/app_adc.c#L319C1-L323C4

If the ADC app is disabled it will not run any set-current, set-rpm etc functions and therefore it stays at this value (and I am not sure if there is any timeout applied, I do not think so...)

It also does not help to override the values to zero first, like this:
```lisp
(if (or (= off 1) (= lock 1))
            (if (not (app-is-output-disabled)) ; Disable output when scooter is turned off
                {
                    (app-adc-override 0 0)
                    (app-adc-override 1 0)
                    (app-disable-output -1)
                }
                
            )
            (if (app-is-output-disabled) ; Enable output when scooter is turned on
                (app-disable-output 0)
            )
)```

What I have not tried yet is setting the current-rel to 0 myself, before disabling the app.

vedderb, 2023-09-23:
(app-disable-output) will stop the app from sending further motor commands, but if the motor is already running it will keep running. You can use set-current 0 afterwards to stop the motor if you do not want it to keep running. Maybe it is better to update app-disable-output so that the motor is released after using it.

1zun4, 2023-09-23:
Yes that would make sense. I also figured out that below the `app_is_output_disabled`, it does the same with `timeout_has_timeout`, so it might keep running there as well?

https://github.com/vedderb/bldc/blob/4e0288d37f203b58acacd735df81ce397319102f/applications/app_adc.c#L319C1-L327C4