vedderb/bldc #656: LispBM Impossible to do "Read Only" with I2C

aelmiger, 2023-09-23:
Hello,

during my testing to make a Wii Nunchuk work with lisp, I got stuck, trying to read the values from the device. I own a logic analyzer and created a minimum working script for an Arduino. When trying to do the same on the vesc and comparing the communication with the logic analyzer, I noticed the following:

To get data from the nunchuck, the following has to be done:
START, 0x00, STOP
wait for a few ms
READ 6 byte

This unfortunately is not possible with the current lisp function for I2C.
The  (i2c-tx-rx addr arrTx optArrRx) function only allows to either send a byte, or to send and immediately read bytes.
There is no option to just read from an adress, which makes it impossible to have the necessary time gap between sending 0x00 and reading the 6 bytes.

vedderb, 2023-09-23:
Can you use an empty list as the tx buffer? E.g.

```clj
(i2c-tx-rx 0x52 '(0))
(sleep 0.005)
(def rx (bufcreate 6))
(i2c-tx-rx 0x52 '() rx)
```

aelmiger, 2023-09-25:
That is something I also tried without success.
Here is the data from the logic analyzer when running it on the vesc:
[image]
[image]

In contrast, this is what is expected when using the arduino:
[image]

When using an empty list, the vesc still sends "START - ADRESS" before reading.

vedderb, 2023-09-26:
A note from the discussion on discord: This was a problem in 6.02 but should be fixed in 6.05. Please let me know if it works there.

aelmiger, 2024-02-06:
Tested and can confirm that on 6.05 it works.
Currently still facing reliability issues, where some i2c messages are not acknowledged, but that could be a hardware issue.

closed 2024-02-06 by aelmiger