vedderb/bldc #658: can-cmd not working reliable

1zun4, 2023-09-28:
`can-cmd` does not work always work, which I use for changing maximum speed, watts and current. This is quite annoying because it causes multiple ESCs being out of sync. I am on BETA 15 and others with BETA 15 experience similiar issues.

My approach is this:
https://github.com/m365fw/vesc_m365_dash/blob/9ac738f598c336d41405ca7faa6cb46920ef0445/g30_dash.lisp#L323-L338

Any ideas? The last thing I could try is combining the can-cmd into a single request, next would be to try BETA 17.

vedderb, 2023-09-28:
I have been working on this one, which is based on flat values:
https://github.com/vedderb/vesc_pkg/tree/main/lib_code_server
It will block and give the result back (or a timeout that can be specified), so you know if it worked. You could also make a wrapper that does some retries. It is much easier to use the code server as you don't need to make a string, you can just quote (or quasi-quote) a value.

Can you give it a try?

1zun4, 2023-10-10:
> I have been working on this one, which is based on flat values: https://github.com/vedderb/vesc_pkg/tree/main/lib_code_server It will block and give the result back (or a timeout that can be [...]

Thank you. I think I can work with this, but so far I am struggling on passing the parametes over. For example:
```
(import "pkg@://vesc_packages/lib_code_server/code_server.vescpkg" 'code-server)
(read-eval-program code-server)

(defun configure-speed(speed watts current)
    {
        (set-param 'max-speed speed)
        (set-param 'l-watt-max watts)
        (set-param 'l-current-max-scale current)
    }
)

(defun set-param (param value)
    {
        (let ((code (str-merge "'(conf-set " "'" (sym2str param) " " (str-from-n value) ")")))
            {
                (print code)
                (print '(conf-set param value))
                (print (list "Input Voltage" (rcode-run 40 0.5 '(get-vin))))
                (print (list "Conf-set" (rcode-run 40 0.5 '(conf-set param value))))
                
                ; I have to pass the variables into the symbol, but how?
            }
        )
    }
)

(configure-speed 20 400 1)
```

I actually have no idea how to do this. 😅

vedderb, 2023-10-10:
It should work well with quasi-quotation. Example:

```clj
(defun set-param (param value)
        (rcode-run 40 0.5 `(conf-set ,param ,value))
)
```

Now you can use things like

```clj
(set-param 'max-speed speed)
(set-param 'l-watt-max watts)
(set-param 'l-current-max-scale current)
```

1zun4, 2023-10-13:
No success so far. :/

[image]

vedderb, 2023-10-15:
My mistake. Can you try this version of set-param?

```clj
(defun set-param (param value)
        (rcode-run 40 0.5 `(conf-set (quote ,param) ,value))
)
```

To check for timeout you can just see if the function returns timeout:

```clj
(if (eq (set-param 'l-watt-max 100000) timeout) (print "Timed out"))
```
That can be used to, for example, make a version of set-param that makes up to 5 retries. Although I'm checking for the result true here as that is what conf-set should return on success.

```clj
(defun set-param (param value)
    (looprange i 0 5 {
            (if (eq (rcode-run 26 0.1 `(conf-set (quote ,param) ,value)) t) (break t))
            false
    })
)
```

vedderb, 2023-10-22:
Another note on this one. Turns out there was a bug in unflatten which would cause random strange behavior and crashes. Should be fixed in the latest beta.

1zun4, 2023-12-08:
So, I have updated to BETA 23 yesterday from BETA 15. Overall, it worked as I was able to setup my scooter.
[image]
[image]

After I setup everything I wanted to upload my script [g30_script](https://github.com/m365fw/vesc_m365_dash/blob/6_05_adc/g30_dash.lisp) and it parsed successfully and runs: Display shows numbers, speed and so on, as well as battery.

BUT
[image]

My VESC running the script disappeared from CAN and I cannot access it anymore.
My guess is that either `can-list-devs` or `can-cmd` is causing CAN to stop working.

Please help. Also seems this issue is present for everyone else too, so it is a software bug.
[image]

vedderb, 2023-12-08:
Can you give me a short example that reproduces that problem that I can try without the hardware you are using?

1zun4, 2023-12-08:
> Can you give me a short example that reproduces that problem that I can try without the hardware you are using?

Not right now as I'm unable to access my VESC anymore. I have to open up the fully sealed scooter :(

1zun4, 2023-12-08:
LUCKY! I was able to recover access to my front ESC by re-flashing the firmware! But this does not help diagnosing the actual issue, as I am not willing to use my scooter for this testing. But I got plenty of MKS_75100_V2_OLD lying around, so I can further test this issue. :) I will let you know. 👍

1zun4, 2023-12-08:
So far I was not able to reproduce the issue of being not able to access the secondary VESC with only using can-cmd, can... etc

But I was able to find another issue, which is `can-list-devs` not reporting anything else than nil
[image]
This is the case for a long period of time, enough that it should already report the secondary VESC.

1zun4, 2023-12-09:
Here you go, that was my attempt at diagnosing the issue: https://youtu.be/fZGosGPmlXM
But like I said, the script is now running and the issue seems to have fixed itself. I cannot tell what caused me losing access even after restarting it multiple times and only being able to reflash the firmware, but I am happy now.

1zun4, 2023-12-09:
So, here you go, this is how I managed to reproduce this issue:
https://www.youtube.com/watch?v=drmO4reIDkg

The code I was talking about for the shutdown is the following:
```lisp
(if (or (= off 1) (= lock 1))
    (if (not (app-is-output-disabled)) ; Disable output when scooter is turned off
        {
            (app-adc-override 0 0)
            (app-adc-override 1 0)
            (app-disable-output -1)
            (set-current 0)
            (loopforeach i (can-list-devs)
                (canset-current i 0)
             )
        }
    )
    (if (app-is-output-disabled) ; Enable output when scooter is turned on
        (app-disable-output 0)
    )
)```

vedderb, 2023-12-09:
There are still a lot of other factors here, such as the bluetooth connection which can be shaky when polling data at the same time as scanning the CAN-bus. There are also many different types of bluetooth modules out there that have different types of problems at higher message rates. I need an example that is as simple as possible, otherwise I have to spent a lot of time reproducing your setup and then I have to remove one factor at a time until I have a simple enough setup so that I can start looking for the problem. If there are hundreds of things going on at the same time interacting with each other it is almost impossible to find the problem.

1zun4, 2023-12-09:
Sorry, that is as much as I can do. I cannot reproduce this issue in another way. Also the factor bluetooth connection is already out, because as you might can see by the first video the CAN communication is entirely broken.

vedderb, 2023-12-09:
That can scan behavior is almost certainly related to bluetooth not keeping up with polling data at the same time. What happens in the first video probably is completely unrelated to the can scan behavior in the second video.

> Also I have noticed that can-list-devs likes to report nil. Should it not report an empty array instead of nil?

An empty list is the same thing as nil.

In that case there is nothing I can do. It is difficult for me to reproduce your exact setup and even in your setup there is no consistent way of reproducing the problem. To make progress I need something smaller, more reproduceable and no TCP/Bluetooth in the middle.

1zun4, 2023-12-10:
So, after removing all can-cmd and can-setcurrent functions calls the script is working fine without the CAN communication getting broken. I give up. I cannot get it to break 100% of the time, sometimes it breaks, sometimes it just works.

```lisp
(print (can-list-devs))

(defun configure-speed(speed watts current fw)
    {
        (set-param 'max-speed speed)
        (set-param 'l-watt-max watts)
        (set-param 'l-current-max-scale current)
        (set-param 'foc-fw-current-max fw)
    }
)

(defun set-param (param value)
    {
        (conf-set param value)
        (loopforeach i (can-list-devs)
            {
                (print i)
                (var cmd (str-merge "(conf-set " "'" (sym2str param) " " (str-from-n value) ")"))
                (print cmd)
                (can-cmd i cmd)
            }
        )
    }
)

(def sport-speed (/ 21 3.6))
(def sport-current 1.0)
(def sport-watts 700)
(def sport-fw 0)

(configure-speed sport-speed sport-watts sport-current sport-fw)
(print "Done!")
```

On my first try:
Console output:
```lisp
Parsing 765 characters
(40)
40
"(conf-set 'max-speed 5.83333)"
```
(CAN lost connection)

But after restarting my VESC and trying it again, I was not able to get the same results. : (

1zun4, 2023-12-11:
I have now finally took your advise to use code-server library and it works!
https://github.com/m365fw/vesc_m365_dash/commit/3508e40db0d0b72f1a3451f4e95f009be378e8e0

So far I have not noticed any desync of configuration, as well as CAN disconnects, but that requires further testing. I took your code example with the 5 retries, because it is very important to have the power limits synced. 👍