vedderb/bldc #663: COMM_GET_IMU_DATA controller_id not set

anscipione, 2023-10-14:
I'm implementing a driver to use vesc inside ros2.

I noticed that the COMM_GET_IMU_DATA
It has a 16-bit mask
line

`uint32_t mask = buffer_get_uint16(data, &ind2);`

but the controller_id will only be obtained with a bit mask
 1 0000 0000 which is not compatible with unit16

line
`if (mask & ((uint32_t)1 << 16)) {`

I think the mask should be:

`uint32_t mask = buffer_get_uint32(data, &ind2);`

is it correct?

vedderb, 2023-10-15:
Good observation!

Here is an attempt at a fix
https://github.com/vedderb/bldc/commit/e0bc115454cf4027d88de0c3692a616a314faa1b

Rather than changing the size of mask I now always add the id to the end. That will keep working with old code that uses the 16-bit mask and actually saves space in the message as one byte for the id is less than two extra bytes for the larger mask on the reply.

anscipione, 2023-10-15:
It seems correct to me. thus the IMU data of several VSCs connected via CANBUS would be recognisable.

closed 2023-10-31 by vedderb